CrealityOfficial/CrealityPrint #248: Creality Print 5.1.6 not running on Arch

basilverthorn, 2024-11-26:
When trying to start the app, it starts and then exists 15 seconds later. Here is what posted back in the terminal when it exits. 4.3 works, but I can not get 5.1.X to start and stay open.

QFont::fromString: Invalid description 'Fira Sans,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'FiraCode Nerd Font Mono,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'Fira Sans,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'Fira Sans,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
X connection to :1 broken (explicit kill or server shutdown).

basilverthorn, 2024-11-27:
So this seems to be an issue running the appimage on Wayland. Although the older 4.3 version runs fine. I reboot and log into an old X11 session it works. What am I missing here, I have been in X in a long time and dont want to go back, love Wayland.

CariSeniorProgrammer, 2024-11-29:
Can not confirm. Have the same error also under X11

basilverthorn, 2024-11-29:
Seems ok under x11. I had install a library but forget which one.  Sent from my iPhone

basilverthorn, 2024-11-29:
Seeing numerous posts on this now on reddit and other arch help forums. Any idea on a fix to get it to run under wayland.

JamesMowery, 2024-11-29:
Also confirming it's not working on Arch Linux. Was considering buying a Creality printer, but seeing the app not work on Arch Linux means I can't buy anything from Creality.

cemox, 2024-12-05:
Same error on Pop_OS! as well

```
Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
XIO:  fatal IO error 2 (No such file or directory) on X server ":1"
      after 771 requests (771 known processed) with 0 events remaining.
[1]    9803 segmentation fault (core dumped)  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
```

mfraser, 2024-12-10:
Same issue on Kubuntu 22.04 with X11.
```
Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
X connection to :0 broken (explicit kill or server shutdown).
Segmentation fault (core dumped)
```

mfraser, 2024-12-10:
> Same issue on Kubuntu 22.04 with X11.
>
> ```
> Initializing StaticPrintConfigsCXBinManager open file size [377984]
> parse binary stl ...face 7558
>
> parse binary stl success...
>
> Computing bounding box... [...]

I even tried extracting the AppImage and running it, same result.

vncsalencar, 2024-12-12:
I have the same error
Running Fedora 41, Nvidia 565.77, Wayland

mfraser, 2024-12-12:
> I have the same error Running Fedora 41, Nvidia 565.77, Wayland

It works if you follow these instructions which I found after commenting on this bug. https://github.com/CrealityOfficial/CrealityPrint/issues/231#issuecomment-2502337437

vncsalencar, 2024-12-12:
@mfraser Unfortunately, does not work for me... Same issues happens (X connection to :0 broken)

schmic, 2025-02-25:
I fixed this by install installing `xwayland`, used in combination with Hyprland.
After restarting my Hyprland session CrealityPrint works fine, even without any env variables set.

closed 2025-06-03 by github-actions[bot]