CrealityOfficial/CrealityPrint #257: Print By Object Error (Object Collision) - K2 Plus

sweenbeen, 2024-12-05:
There is a bug or chance to enhance Creality Print v 5.1.0 - When selecting Other/Special Mode/Print Sequence --> Print by Object - the slicer should determine the print order by the smallest object first - then navigate to printing the tallest object last. This is not occurring - I have been presented with Collision warnings when items are significantly different in height; However, the objects are sufficiently spaced to avoid collisions - As a result of this warning - the slicer prevents you from slicing the file. 

[image]
[image]

sweenbeen, 2024-12-05:
[image]

In case there is concern with color sequencing too.  I re-organized to align the colors next to one another.

nikil511, 2024-12-29:
+1 
also print order seems to have no effect, so basicly imposible to print "by object" two different objects
[image]

closed 2025-04-28 by github-actions[bot]