CrealityOfficial/CrealityPrint #262: 5.1.7 New error: Starts and Exits

basilverthorn, 2024-12-15:
Getting a new error with 5.17 pre-release.

./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage
Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":1"
after 12567 requests (12564 known processed) with 0 events remaining.
Fatal glibc error: tpp.c:83 (__pthread_tpp_change_priority): assertion failed: new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)

butfar, 2024-12-18:
The problem with the interaction of the NVideo Geforce 1650 video card in the latest release of the images program for MintLinux has not been resolved. The problem is that the program opens and closes immediately, and if you roll back the drivers on Linux, the program works.

wbvczar, 2024-12-19:
Or you can keep the latest Nvidia drivers and switch back to X11.  The Creality Print software doesn't like Nvidia drivers with Wayland.

ralexander87, 2024-12-19:
As I had the same issue. after some time, fixed .appimage
I'm on Arch with AMD on CPU and nVidia as 2nd. Using Hyprland. 
Since the CP is XORG (IIRC), I could not run it on Hyprland for some reason. KDE, GNOME, and all other CP worked in any way run.
original I did go with
`env __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=amd ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage`

It works also:
`env __GLX_VENDOR_LIBRARY_NAME=amd ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage`

The last part I didn't have as an error
`Fatal glibc error: tpp.c:83 (__pthread_tpp_change_priority): assertion failed: new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)` so I'm not sure if this will fix the problem you have.
On CP discord, there is talk of the same.

basilverthorn, 2024-12-26:
No luck with the above either or the post on adding the following to ENV:

WEBKIT_DISABLE_DMABUF_RENDERER=1

closed 2025-04-28 by github-actions[bot]