CrealityOfficial/CrealityPrint #265: Source code missing - GPL violation?

arosl, 2024-12-20:
Hi Creality,

I’m hoping this is just a simple oversight on my part or a temporary issue with your release process, but I’m having trouble finding the source code for any version newer than 5.0.3.

On your release pages, the attached source.zip files only contain the license and a readme.md, with no actual source code. The link on your website redirects back to the same page, and I haven’t been able to locate the updated code elsewhere.

While it’s not strictly required, it would be greatly appreciated, and in keeping with the spirit of open source, if you could push the complete source code to the repository rather than only throwing the code over the wall each release. At the very least, please ensure that the full code is included for each new release.

Thank you.

arosl, 2024-12-20:
@hemiao2023 @CrealityTech

hemiao2023, 2024-12-20:
The source code has been updated to 5.1.4, and the latest source code requires some internal review before it can be released. @arosl

arosl, 2025-01-02:
Creality Team, @hemiao2023 @CrealityTech 

Your Creality Slicer software is distributed under the **GNU General Public License v3 (GPL v3)**. Under this license, **any** party distributing a binary or executable version of GPL-licensed software must **simultaneously**:

1. Provide the **complete corresponding source code**, or  
2. Offer a valid, written, legally binding option to obtain the source code at no more than a nominal fee for distribution.

Recent updates beyond version 5.1.4, including the 6.0 release, are available in binary form without the **matching, up-to-date source code**. Stating that the latest source code “requires internal review” before release does **not** meet the immediate publication requirement of the GPL. **All** binary-only releases that lack their corresponding source code are in breach of the GPL v3 license terms.

Furthermore, under GPL v3, “complete corresponding source code” includes **everything necessary** to build the software, such as scripts or configuration files. An incomplete or partial source distribution is not sufficient.

To bring Creality Slicer back into compliance, the following steps are requested:

1. **Cease distribution** of any slicer binaries (including versions between 5.1.4 and 6.0) that do not have matching and complete source code publicly available.  
2. **Immediately publish** the full source code, including all build scripts and configuration files required to recreate **each** existing release.  
3. **Ensure future releases** fully comply with GPL v3: either publish the exact, complete source code in tandem with the binary releases or provide a valid written offer at the time of publication.

By following these steps, you respect both your obligations under GPL v3 and the rights of the wider community to study, modify, and share improvements to your software. Thank you for your prompt attention to this matter.

hemiao2023, 2025-02-11:
All source code has been uploaded to the master branch