CrealityOfficial/CrealityPrint #267: App chrashes on Linux Mint

SKajto, 2024-12-25:
The app chrashes on start on Linux Mint with the following output:
`Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
XIO:  fatal IO error 2 (No such file or directory) on X server ":0"
      after 734 requests (734 known processed) with 0 events remaining.
Segmentation fault (core dumped)
`
Tested with [V5.1.7](https://github.com/CrealityOfficial/CrealityPrint/releases/tag/v5.1.7) and [v5.1.6](https://github.com/CrealityOfficial/CrealityPrint/releases/tag/v5.1.6) with the same result.

System configuration:
OS: Linux Mint 22 Cinnamon 6.2.9
Kernel: 6.8.0-50-generic
CPU: AMD Ryzen 7 5700X 8-Core Processor × 8
RAM: 125.7 GiB
Hard Drives: 8082.3 GB
GPU: NVIDIA Corporation AD106 [GeForce RTX 4060 Ti]
Display Server: X11

SKajto, 2024-12-25:
The workaround as mentioned in https://github.com/CrealityOfficial/CrealityPrint/issues/248 is working for me:
```
env __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage
```

basilverthorn, 2024-12-26:
No glory here with this or WEBKIT_DISABLE_DMABUF_RENDERER=1 as an option.

saperlot, 2024-12-26:
It does not crash on my linux mint!

jyriturkia, 2025-02-05:
openSUSE Tumbleweed
KDE Plasma 6.2.5
Kernel 6.13.1
Wayland
Nvidia drivers 570.86.16

same issue.

> ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage
Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
XIO:  fatal IO error 2 (No such file or directory) on X server ":1"
      after 961 requests (961 known processed) with 0 events remaining.
Segmentation fault (core dumped)

wen422, 2025-05-15:
Creality bot: This issue was closed because it has been inactive for 7 days since being marked as stale.

closed 2025-05-15 by wen422