CrealityOfficial/CrealityPrint #272: Creality Print 6.0 (Windows) fails to add Sonic Pad

mdrjr, 2024-12-30:
Creality Print 6.0 won't find a sonic pad on network.
When trying to add the sonic pad manually if I add the ip address and set Creality as the type it says device not found.

Sonic Pad is on the network, IP is accessible by browser and no Windows Firewall (disabled)

rvoyum, 2025-01-01:
I'm having the same problem with my Sonic Pad. 

CP 6.0 finds my K1 Max, but not other printers or my Sonic pad with a CR-10 Smart Pro attached.
The latest version of CP 5.xx finds both pad and printers without any problem.

CP v6.0-1078 only finds Flagship printers that are connected to the network.

If I try to enter the IP address manually, I only get a message that the slicer does not exist.

JimD1445, 2025-01-04:
Having the same issue with Print 6.0-1078 not seeing the sonic pad. Version 5.1 finds the Sonic Pad and my other printer on the network  (Ender 3 KE)

alekseyzal, 2025-01-05:
попробуйте отключить файервол, тогда всё найдётся

orca442, 2025-01-07:
I got it working by "Manual Add" adding "Other Printer" then inputting the ip address.  This will show the Klipper web interface. Not ideal, but will work.

AnthonyDewhirst, 2025-02-03:
I am also having the same issue on Mac

brvwvde, 2025-02-09:
> попробуйте отключить файервол, тогда всё найдётся
> 
> Проблемы с сетевыми подключениями оказались просты, теперь Creality Print 6 , использует процесс "msedgewebview2.exe", пришлось открыть доступ в файерволе этому приложению к сети и всё заработало....

все оказалось не так просто, добавил в исключения это приложение, все еще The device does not exist в Creality Print

wen422, 2025-05-11:
Creality bot: this issue is stale because it has been open for 90 days with no activity.

wen422, 2025-05-18:
Creality bot: This issue was closed because it has been inactive for 7 days since being marked as stale.

closed 2025-05-18 by wen422