CrealityOfficial/CrealityPrint #273: Creality Print 6.0 - Sign In

WyldCFH, 2024-12-30:
Hello,

Using CP 6.0 on Windows, I cannot log in. I hit the login button, it goes to the id.creality page, log in there successfully but no change in CP 6.0.
Leaving the id.creality page open, signed in, and click the same sign in button takes me to another page for id.creality that is not signed in.

WyldCFH, 2024-12-31:
A Reddit user posted there is supposed to be some type of dialog to allow the page and CP to connect.
I tried with Chrome, Firefox, Edge and Opera and got no such dialog or option.
Edge and Firefox I had never logged into anything Creality before, so browser cache is not the issue

Lupu5R3x, 2025-01-01:
Same here, no login in v6. And one should think that if we already was logged into v5, then v6 should pick that login up as pr default.
Same with all the custom settings, they should also be loaded by default, without the user needs to move files etc.

WyldCFH, 2025-01-03:
6.0.0.1078 seems to, for me at least, to fix the issue with the web page popup working now. 
I was successfully able to log in.

alekseyzal, 2025-01-05:
Проблемы с сетевыми подключениями оказались просты, теперь Creality Print 6 , использует процесс "msedgewebview2.exe", пришлось открыть доступ этому приложению к сети и всё заработало....

ppb85, 2025-01-12:
If you use any sort of antivirus that has add and pop up blocked or pop up blocker in web browser you will have issues. I've just excluded creality web and works perfect now on chrome.

Lupu5R3x, 2025-01-19:
With 6.0.1.1375 I still only have a Untitled titled, where the sig-in info should be. And I didn't had to disable any thing in v5.1 to sign in. So there should be no reason to do so in v6 either...

jimfromarizona, 2025-01-26:
I have the same problem

macOS 15.2
Safari 18.2
Creality Print 6.0.2.1576

nugisn, 2025-02-23:
I FEEL SO STUPID!

Was searching the web for the answer thinking Creality was the problem, then realised what caused a similar problem on another website. 

AD-  AND POPUP BLOCKERS. Disable them on creality website, and should work fine 😄

widgit, 2025-03-01:
Hi,
Confirmed not working with Chrome with creality.com in the allow list for popup.

Windows 11 Home - Version	10.0.22631 Build 22631
Chrome -  132.0.6834.197

DeepflashX, 2025-04-05:
I also wonder why I have to login everytime again once opening up crealityprint. Also I never get the login in the current crealityprint instance but only in a second one which then opens, super weird. I am on 6.1.0.2041 on windows 11

closed 2025-07-10 by Dai-112100

Lupu5R3x, 2025-07-10:
I can confirm that in 6.2 I can login with no issues. 👍

alekseyzal, 2025-07-10:
В версии 6.2 входит без проблем, но проблема другая, 6.2 не делает заполнение пустоты в деталей, приходится ставить 6.1