CrealityOfficial/CrealityPrint #274: Switch from 5 to 6

C-Noob404, 2024-12-31:
Hi all,
can someone explain how we can easily switch from creality print 5 to 6? Which folders need to be copied to save all manual settings? Cura for example stores everything in the cloud, so its easy to backup and restore. How is the workflow for this one?

Thank you very much in advance!

GraymanV, 2025-01-05:
In 5.1 in the top menu press File -> Export -> Export presets.
Then in 6.0: File -> Import -> Import configs and just open zip file, that you get before.

irudi, 2025-04-24:
I've tried - but does not work for me (with Creality Print 6.1.1 beta on macOS 15.4). The imported settings are not visible.

wen422, 2025-05-23:
Upgrading from CP4.x to CP5.x indeed requires recreating presets. However, during the initial installation and launch of CP6.x, a prompt will appear asking, "Migrate user presets from version 5.x?", Clicking ‌"Confirm‌" will initiate automatic migration. You can also manually import presets via ‌"File → Import → Import 5.x Presets‌", as illustrated below:

[image]

You can also refer to this video:https://www.youtube.com/watch?v=CMqt-pm_9zw

wen422, 2025-05-28:
> Unfortunately this does not work for me! Maybe it's a bug in the macOS version?
> 
> Here my presets in 5.17
> 
> [image]
> 
> If I try to import from 6.1.1
> 
> [image]

Could you provide your preset files?

irudi, 2025-05-28:
> Could you provide your preset files?

Sure. How to collect them?