CrealityOfficial/CrealityPrint #275: Bad CP 6.0 GUI on 4K

dirk-niemeier, 2024-12-31:
Please try 6.0.0.1074 on a 4K System (Win 11, Nvidia):

- Top Icons by far too large.
- Fonts appear disproportioned on some menus and dialog.

This can partially be fixed in the compatibilities settings, however areas of the display will become blurry and the separator between the print settings and the graphical area in prepare mode won't work properly.

aliasguru, 2025-01-01:
Same issue here. Sizes of GUI elements are all over the place, check start screen here:

[image]

Quite a few elements become unreadable:

[image]

On top of that, 3D viewport performance is really poor on my system compared to CP 5.1, as if the viewport was rendered on CPU.
Will stick with 5.1 until a fixed release comes out

black8592, 2025-01-03:
Send file screen also bugged for me: 
[image]

closed 2025-04-28 by github-actions[bot]