CrealityOfficial/CrealityPrint #280: Creality Print 6.x forgets what filament type was used and defaults to Hyper PLA

zme-ul, 2025-01-02:
in version 5.1.x Print will aways remember the last type of used filament, Print 6.x reverts back to Hyper PLA every time the program is started

CalamitouisJ, 2025-01-03:
Seconding this, just updated from 5.1 and 6 is reverting to default filament and default process every startup

Rockj71, 2025-01-04:
Stesso problema con la versione 6.0

saperlot, 2025-01-06:
i have noticed that at some point the type of the fillament had changed to another one. There are definately some errors in the system.

zme-ul, 2025-01-10:
I can confirm Print 6.0.1.1375 remembers filament type after exiting

closed 2025-01-10 by zme-ul

Rockj71, 2025-01-11:
> Posso confermare che Print 6.0.1.1375 ricorda il tipo di filamento dopo l'uscita

cosa vuol dire dopo l'uscita? ho aggiornato alla 6.0.1.1375 ma il problema persiste