CrealityOfficial/CrealityPrint #283: CP6 Camera Not Working

frankt2012, 2025-01-03:
Camera works fine in CP5 but not in either releases of CP6.

kkeonline, 2025-01-04:
> Camera works fine in CP5 but not in either releases of CP6.

My nebula cam work fine in CP6

frankt2012, 2025-01-04:
> > Camera works fine in CP5 but not in either releases of CP6.
> 
> My nebula cam work fine in CP6 (Ender 3 V3 KE printer) Did you update printer firmware to the latest version?
> 
> ![2025-01-05_012141](https://private-user-images.githubusercontent.com/30925725/400157896-873c003f-19af-4b3d-aa39-14b8e0821145.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwMTUzNjcsIm5iZiI6MTczNjAxNTA2NywicGF0aCI6Ii8zMDkyNTcyNS80MDAxNTc4OTYtODczYzAwM2YtMTlhZi00YjNkLWFhMzktMTRiOGUwODIxMTQ1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA0VDE4MjQyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRmNzAzNDkxZDg5MjZkODQ0YTY1ZjIyZTNlNTQ4MDhlZWRmNGFiNWVmYjU5YzVkMGNmYmI1OGEyNGVmNGM1NjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.F7Re-tsXpPZYD67IXUuonqv82Azwd27xsgEPGK2PdYc)

I am on the latest K2 firmware. Just updated last night to 1.1.1.7. Camera works fine in CP5 and if I use Opera and not a Chrome based browser I can access through the IP.

kkeonline, 2025-01-04:
> I am on the latest K2 firmware. Just updated last night to 1.1.1.7. Camera works fine in CP5 and if I use Opera and not a Chrome based browser I can access through the IP.

I'm leaving Creality Print and opt-in Orca Slicer
Creality print release new version too fast without correct bug(ssss) in old version and new version alway have new bug(sss)
Orca Slicer now support wireless print to creality printer without need to root and install fluidd, it so stable and less bug.

saperlot, 2025-01-06:
i think mine was also not working on the first start. but now without issues, used already few times..

frankt2012, 2025-01-09:
Camera still not working over LAN on the new 6.0.1 version just released. Even did a clean install and I just get reconnecting.

assafavi, 2025-01-10:
Camera still not working over LAN on the new 6.0.1 version just released. I just get reconnecting.
But Camera still works on last version of CP5

frankt2012, 2025-01-10:
> Camera still not working over LAN on the new 6.0.1 version just released. I just get reconnecting. But Camera still works on last version of CP5

Yep still the same issue for me to.

Avrution, 2025-01-17:
Can confirm.  Just setup my K2 for lan use only and no camera.

assafavi, 2025-01-17:
Yes, just setup for LAN use.
(Win 11 pro,  CP 6.0.1.1375) No Camera. I can send gcode to K2 but can't Print. It does not recognize the CFS

FYI - Camera and print function are still working in CP5

Thanks

frankt2012, 2025-01-17:
Camera is finally working now in 6.0.2

closed 2025-01-17 by frankt2012