CrealityOfficial/CrealityPrint #288: CrealityPrint V6.0.1 login

sevagjb, 2025-01-09:
i cant login with my Creality account
still the same problem sins first release

jedoCodes, 2025-01-23:
I also can't log in with my Creality account.    
Safari tells me that the URL is incorrect.

macOS 15
Safari 18
Creality.Print 6.0.2

jimfromarizona, 2025-01-26:
I have the same problem

macOS 15.2
Safari 18.2
Creality Print 6.0.2.1576

sevagjb, 2025-01-28:
Is the maintenance team aware of these issues? We are still unable to log in.

somenkovnikita, 2025-02-28:
Tested in v6.0.4 — everything works perfectly. The issue was most likely here:

```diff
-      <dict>
-        <key>CFBundleURLName</key>
-        <string>CrealityPrint Downloads</string>
-        <key>CFBundleURLSchemes</key>
-        <array>
-          <string>crealityprint</string>
-        </array>
-      </dict>
-    </array>
-  <array>
+    <dict>  
+       <key>CFBundleURLSchemes</key>  
+        <array>  
+           <string>CrealityPrintLink</string>  
+       </array>  
+      <key>CFBundleURLName</key>  
+      <string>com.creality.crealityprint</string>  
+  </dict>
+ </array>
```

sevagjb, 2025-02-28:
No, I solve the problem with a different approach. You need to log in using a different browser to link your account with the slicer.