CrealityOfficial/CrealityPrint #299: Feature Request: Option Zoom to / rotate around selected object, mouse cursor or screen center

ScruffR, 2025-01-15:
It would be nice to have a quick switch between ways to rotate and zoom, depending on momentary needs.
This could be achieved by using the modifier keys <kbd>Shift</kbd>, <kbd>Control</kbd><kbd>Alt</kbd>

Lancelight, 2025-01-22:
This was a MAJOR complaint of mine as well but it went completely ignored for a while.  They silently re-enabled the zoom to cursor option where you dont have to move the plate to zoom into specific areas of the plate in the latest version which came out today.  You point your mouse on the point you want to zoom to which is better than the shift/control/alt method anyway.  The problem is that it's not on by default.  You have to go to "Preferences" (the little wheel in the toolbar) to enable it,  it's called Zoom to cursor position.

ScruffR, 2025-01-22:
That's a great news. I would not have thought to look into Preferences again, since there was no such option and the addition was not mentioned in the changelog either.

closed 2025-05-01 by github-actions[bot]