CrealityOfficial/CrealityPrint #311: Missing AppImage or other binaris for Linux

chrisruffalo, 2025-01-25:
Please publish the Linux builds. Thanks.

Liionboy, 2025-01-26:
They don't care about people who use linux, same for me i need linux version but i change to Orca because they don't like linux user

jaredcrean, 2025-01-27:
A Linux build would be much appreciated.

mcwalter33, 2025-01-28:
Is there any eta? 
Driving users to other products

Liionboy, 2025-02-01:
anyoone can answer to us? why nobody from dev don't answer to this question i don't know

chrisruffalo, 2025-02-02:
It took a few releases from 6.0 to add the mac image to the release _after_ someone asked for it. I think showing how many people are looking for this is helpful for prioritizing.

jaredcrean, 2025-02-03:
I would be nice to get a docker container setup with the build environment. That way we could build it our selfs. I tried to get it building but gave up on trying to figure it out.

spinlokd, 2025-02-05:
Was able to build on Fedora 41
steps:

1. install dependencies as per README.md: libmspack-dev libgstreamerd-3-dev libsecret-1-dev libwebkit2gtk-4.0-dev libosmesa6-dev libssl-dev libcurl4-openssl-dev eglexternalplatform-dev libudev-dev libdbus-1-dev extra-cmake-modules libgtk2.0-dev libglew-dev libudev-dev libdbus-1-dev cmake git texinfo
2. install libquadmath-devel
3. Open CMakeLists.txt and add the following line: `add_compile_options(-Wno-error=template-id-cdtor)`. I added it at line 153, where i saw other uses of `add_compile_options` function
4. `sudo ./BuildLinux.sh -u`
5. `./BuildLinux.sh -dsi`

It starts but then segfaults. might be due to Wayland??

Liionboy, 2025-02-05:
Can you upload here appimage to test on ubuntu?

spinlokd, 2025-02-05:
> Can you upload here appimage to test on ubuntu?

I get "failed to upload" message. maybe because it's too big?

Liionboy, 2025-02-05:
> > Can you upload here appimage to test on ubuntu?
> 
> I get "failed to upload" message. maybe because it's too big?

upload somewhere and put here link, i want to test it, i can't compile on Fedora 41 i have same error

Liionboy, 2025-02-05:
```
1/16] Performing patch step for 'dep_OCCT'
FAILED: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch /root/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch 
cd /root/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT && git apply --directory deps/build/dep_OCCT-prefix/src/dep_OCCT --verbose --ignore-space-change --whitespace=fix /root/CrealityPrint-6.0.2/deps/OCCT/0001-OCCT-fix.patch && /usr/bin/cmake -E touch /root/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory
[2/16] Performing patch step for 'dep_OpenCV'
FAILED: dep_OpenCV-prefix/src/dep_OpenCV-stamp/dep_OpenCV-patch /root/CrealityPrint-6.0.2/deps/build/dep_OpenCV-prefix/src/dep_OpenCV-stamp/dep_OpenCV-patch 
cd /root/CrealityPrint-6.0.2/deps/build/dep_OpenCV-prefix/src/dep_OpenCV && git apply --directory deps/build/dep_OpenCV-prefix/src/dep_OpenCV --verbose --ignore-space-change --whitespace=fix /root/CrealityPrint-6.0.2/deps/OpenCV/0001-vs2022.patch && /usr/bin/cmake -E touch /root/CrealityPrint-6.0.2/deps/build/dep_OpenCV-prefix/src/dep_OpenCV-stamp/dep_OpenCV-patch
Checking patch deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/templates/OpenCVConfig.root-WIN32.cmake.in...
error: deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/templates/OpenCVConfig.root-WIN32.cmake.in: No such file or directory
Checking patch deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/OpenCVDetectCXXCompiler.cmake...
error: deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/OpenCVDetectCXXCompiler.cmake: No such file or directory
[3/16] Performing update step for 'dep_wxWidgets'
-- Fetching latest from the remote origin
ninja: build stopped: subcommand failed.
```

Liionboy, 2025-02-05:
> Was able to build on Fedora 41 steps:
>
> 1. install dependencies as per README.md: libmspack-dev libgstreamerd-3-dev libsecret-1-dev libwebkit2gtk-4.0-dev libosmesa6-dev libssl-dev [...]

did you clone from git or download Source Code? Because with Source Code i have same error at compile like on ubuntu

chrisruffalo, 2025-02-05:
I'm running ``.

I don't really know C or any of the C build ecosystem but I jumped through a few hoops. The first was running the build update command:
```
[]$ chmod +x BuildLinux.sh
[]$ sudo ./BuildLinux.sh -u
```
That failed and I had to update the `linux.d/debian` file to have version 4.1 of the libwebkit2-gtk library. Then I ran it again and it completed successfully.

After that I am able to run `./BuildLinux.sh -dsir` and I get the following error:
```
[1/53] Performing patch step for 'dep_OCCT'
FAILED: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch /home/chris/src/creality-print/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch 
cd /home/chris/src/creality-print/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT && git apply --directory deps/build/dep_OCCT-prefix/src/dep_OCCT --verbose --ignore-space-change --whitespace=fix /home/chris/src/creality-print/CrealityPrint-6.0.2/deps/OCCT/0001-OCCT-fix.patch && /usr/bin/cmake -E touch /home/chris/src/creality-print/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory
[2/53] Performing patch step for 'dep_OpenCV'
FAILED: dep_OpenCV-prefix/src/dep_OpenCV-stamp/dep_OpenCV-patch /home/chris/src/creality-print/CrealityPrint-6.0.2/deps/build/dep_OpenCV-prefix/src/dep_OpenCV-stamp/dep_OpenCV-patch 
cd /home/chris/src/creality-print/CrealityPrint-6.0.2/deps/build/dep_OpenCV-prefix/src/dep_OpenCV && git apply --directory deps/build/dep_OpenCV-prefix/src/dep_OpenCV --verbose --ignore-space-change --whitespace=fix /home/chris/src/creality-print/CrealityPrint-6.0.2/deps/OpenCV/0001-vs2022.patch && /usr/bin/cmake -E touch /home/chris/src/creality-print/CrealityPrint-6.0.2/deps/build/dep_OpenCV-prefix/src/dep_OpenCV-stamp/dep_OpenCV-patch
Checking patch deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/templates/OpenCVConfig.root-WIN32.cmake.in...
error: deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/templates/OpenCVConfig.root-WIN32.cmake.in: No such file or directory
Checking patch deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/OpenCVDetectCXXCompiler.cmake...
error: deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/OpenCVDetectCXXCompiler.cmake: No such file or directory
```

Those files exist so it's probably got a prefix wrong somewhere and I'm about at my limit with what I can understand and attempt to fix. There's a Dockerfile and I ran the following:
```
[]$ podman build --rm -t creality-print:6.0.2 .
```
And that fails too.

spinlokd, 2025-02-06:
Try this: https://github.com/spinlokd/CrealityPrint/releases/download/v6.0.2/CrealityPrint_V6.0.0_Alpha.AppImage
I forked the repo :)

Liionboy, 2025-02-06:
So is the same error when i try to run like on Ubuntu:

```
adrian@adrian-huawei-d16:~/Downloads$ ./CrealityPrint_V6.0.0_Alpha.AppImage 
[2025-02-06 08:31:46.793579] [0x00007681811d6400] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(process:883886): GLib-GObject-CRITICAL **: 08:31:47.063: invalid (NULL) pointer instance

(process:883886): GLib-GObject-CRITICAL **: 08:31:47.063: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

spinlokd, 2025-02-06:
> So is the same error when i try to run like on Ubuntu:
>
> ```
> adrian@adrian-huawei-d16:~/Downloads$ ./CrealityPrint_V6.0.0_Alpha.AppImage
> [2025-02-06 08:31:46.793579] [0x00007681811d6400] [trace] [...]

These are the errors i was seeing on my env. I chalked them up to kde/wayland/nvidia

Liionboy, 2025-02-06:
[image]

I don't have nvidia or something like that , i am using Linux Mint but again thwy don't want to help use

spinlokd, 2025-02-07:
I believe this issue is resolved in BambuStudio by bumping webkit2gtk dep from webkit2gtk-4.0 to 4.1. I guess once Orca pulls in those changes and CrealityPrint pulls in latest Orca, it'll be fixed here as well

Liionboy, 2025-02-07:
In the source code right because on my linux mint i have that version:

```
adrian@adrian-huawei-d16:~$ sudo apt install libwebkit2gtk-4.1-dev
[sudo] password for adrian:          
Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 684704 (synaptic)      
Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 684704 (synaptic)      
Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 684704 (synaptic)      
Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 684704 (synaptic)      
Waiting for cache lock: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 684704 (synaptic)      
Reading package lists... Done                                                                                  
Building dependency tree... Done
Reading state information... Done
libwebkit2gtk-4.1-dev is already the newest version (2.46.5-0ubuntu0.24.04.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

Liionboy, 2025-02-07:
after i remove the folder /home/adrian/.config/Creality i have a new error
```
(CrealityPrint:884901): Gtk-CRITICAL : 01:39:57.462: gtk_widget_set_size_request: assertion 'width >= -1' failed

(CrealityPrint:884901): Gtk-CRITICAL : 01:39:57.462: gtk_widget_set_size_request: assertion 'width >= -1' failed

(CrealityPrint:884901): Gtk-CRITICAL : 01:39:57.462: gtk_widget_set_size_request: assertion 'width >= -1' failed

(CrealityPrint:884901): Gtk-CRITICAL : 01:39:57.462: gtk_widget_set_size_request: assertion 'width >= -1' failed
Segmentation fault (core dumped)
```

Liionboy, 2025-02-08:
Another debug log, and i see there are missing some library:

`adrian@adrian-huawei-d16:~/Downloads$ gdb CrealityPrint_V6.0.0_Alpha.AppImage 
GNU gdb (Ubuntu 15.0.50.20240403-0ubuntu1) 15.0.50.20240403-git
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from CrealityPrint_V6.0.0_Alpha.AppImage...
(No debugging symbols found in CrealityPrint_V6.0.0_Alpha.AppImage)
(gdb) run
Starting program: /home/adrian/Downloads/CrealityPrint_V6.0.0_Alpha.AppImage 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 946216]
process 946213 is executing new program: /usr/bin/bash
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libtinfo.so.6
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 946222]
process 946213 is executing new program: /tmp/.mount_CrealiYZAUIj/bin/CrealityPrint
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgdk-3.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgobject-2.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libglib-2.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgstvideo-1.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libbrotlidec.so.1
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgmodule-2.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/liborc-0.4.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libicui18n.so.70
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libicuuc.so.70
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libxslt.so.1
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libwoff2dec.so.1.0.2
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgstapp-1.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgstpbutils-1.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgstaudio-1.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgsttag-1.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgstfft-1.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libenchant-2.so.2
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libbrotlicommon.so.1
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libcap.so.2
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libwoff2common.so.1.0.2
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgstallocators-1.0.so.0
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libtinfo.so.6
[2025-02-08 13:12:18.848819] [0x00007fffe2d89400] [trace]   Initializing StaticPrintConfigs
[New Thread 0x7fffe02006c0 (LWP 946233)]
[New Thread 0x7fffd7e006c0 (LWP 946234)]
[Thread 0x7fffe02006c0 (LWP 946233) exited]
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
warning: could not find '.gnu_debugaltlink' file for /usr/lib/x86_64-linux-gnu/gtk-3.0/modules/libxapp-gtk3-module.so
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libxapp.so.1
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libgnomekbdui.so.8
[... 149 lines cut ...]
(CrealityPrint:946213): Gtk-CRITICAL **: 13:12:23.650: gtk_widget_set_size_request: assertion 'width >= -1' failed

(CrealityPrint:946213): Gtk-CRITICAL **: 13:12:23.650: gtk_widget_set_size_request: assertion 'width >= -1' failed

(CrealityPrint:946213): Gtk-CRITICAL **: 13:12:23.650: gtk_widget_set_size_request: assertion 'width >= -1' failed

(CrealityPrint:946213): Gtk-CRITICAL **: 13:12:23.650: gtk_widget_set_size_request: assertion 'width >= -1' failed

[... cut ...]
#5  0x000000000447e169 in wxWindowBase::DestroyChildren() ()
#6  0x0000000004564630 in wxWindow::~wxWindow() ()
#7  0x00000000004e7497 in Slic3r::GUI::MainFrame::MainFrame() [clone .cold] ()
#8  0x0000000001658d95 in Slic3r::GUI::GUI_App::on_init_inner() ()
#9  0x0000000001659979 in Slic3r::GUI::GUI_App::OnInit() ()
#10 0x00000000046c2343 in wxEntry(int&, wchar_t**) ()
#11 0x000000000161123b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#12 0x000000000099825f in Slic3r::CLI::run(int, char**) ()
#13 0x00000000008b9c33 in main ()`

wen422, 2025-05-10:
Creality bot: this issue is stale because it has been open for 90 days with no activity.

ymettier, 2025-05-10:
v6.1.1 published and still no appimage or any official linux build.

> Please publish the Linux builds. Thanks.

Please...

wen422, 2025-06-03:
We have released v6.1.2, including the Linux version.
Ubuntu：https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.2/CrealityPrint-V6.1.2.2458-x86_64-Release.AppImage
Ubuntu24.04：https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.2/CrealityPrint_Ubuntu2404-V6.1.2.2458-x86_64-Release.AppImage

closed 2025-06-03 by wen422