CrealityOfficial/CrealityPrint #315: Print V6.0.2 filament not saved upon exit

zme-ul, 2025-01-29:
just like reported previously, https://github.com/CrealityOfficial/CrealityPrint/issues/280 the issue is back with **v6.0.2**, it defaults to Hyper PLA

the issue was fixed in version 6.0.1

zme-ul, 2025-02-15:
this issue persists with v6.0.3.1591 pre-release build

Sdsyc, 2025-02-17:
The same thing happens to me, it persists with the **CrealityPrint_6.0.3.1591_Release** version, when exiting the selected custom configuration is not saved, **it always starts with the system configuration and with Hyper PLA.**

zme-ul, 2025-02-28:
the issue seems fixed with build v6.0.4 Pre-release

altmenorg, 2025-05-22:
Still an issue in 6.1.1...

zme-ul, 2025-05-22:
> Still an issue in 6.1.1...

no, I cannot confirm it - it works fine on mine

closed 2025-07-10 by Dai-112100