CrealityOfficial/CrealityPrint #319: Creality Print 6.0.2.1574 intermittently fails to initialise Wi-Fi connected printer

Wibblepuppy, 2025-01-31:
**Problem:** After sending a successful print job, change the STL on the plate via right click option (in order to reuse settings), slice it and print. Intermittently nothing outputs.

**Symptoms:** The printer status screen looks normal (printer idle, temperatures on show) then Send to printer takes an extended time to finish and for the upload bar to complete. This slow upload condition is indicative that a failure is inevitable. The view changes to printer status screen and conditions are blank. Temperatures are shown as "-" with no numeric. The graphic representing the print job remains as the default cube icon and the printer does not initialise.

**Solution:** Requires closing Creality Print application and to restart the print job either from scratch or via a saved project.

MastaQ1, 2025-02-03:
I am seeing the same issue. I am sometimes able to send the print from the home of the details page by checking off my printer then clicking the send to printer button on the left where it shows the model. Also slice the model then save the file as gcode and upload from web interface via IP.

Dai-112100, 2025-02-12:
Hello，I am the Creality PM of slicing software. 

Thank you for your feedback. We highly value the issue you have raised.

We have analyzed the issue and found that it may also be related to the printer. 
Please provide the printer model and firmware version to help us quickly identify the problem.
It would be best if you could also provide the 3MF file of the software project related to this issue and the Gcode file.

1. printer model
2.  firmware version
3. The 3mf file in question
4. The Gcode file in question

Dai-112100, 2025-02-12:
Hello I am the Creality PM of Creality Print

Thank you for your feedback. We highly value the issue you have raised.

We have analyzed the issue and found that it may also be related to the printer. 
Please provide the printer model and firmware version to help us quickly identify the problem.
It would be best if you could also provide the 3MF file of the software project related to this issue and the Gcode file.

1. Printer model
2. Firmware version
3. The 3MF file in problem
4. The Gcode file in problem

> **Problem:** After sending a successful print job, change the STL on the plate via right click option (in order to reuse settings), slice it and print. Intermittently nothing outputs.
>
> **Symptoms:** [...]

eduneris71, 2025-03-15:
Hi, I have the same problem in my K1-max when I send the gcode by wifi. It takes a long time to upload. Sometimes it starts the printer, sometimes get upload failed. The 3.mf file is only 1.5MB. This problem is happening since version 6.03. I will try to come back to v.6.02 or older versions to avoid this problem.

eduneris71, 2025-03-15:
Just to update. After reinstall 6.00 release, wifi is working. I will keep this release until the newest version is wortking

closed 2025-06-21 by github-actions[bot]