CrealityOfficial/CrealityPrint #321: Several build-dependencies force taboo breach (root permission), what a terrible failure ...

rubienr, 2025-02-01:
# What a Terrible Failure (WTF)

On Linux, several `dep_*` dependencies <ins>__require root permission__</ins> in order to compile the prerequisite targets for the OrcaSlicer, since they want to install themselves system wide.
[image]

As a developer i will not go into details and list reasons why the community isn't going to accept that circumstance,
and won't give root permissions to the building process that fetches a pot of dependencies from sources other than the Linux-distributors' repository and installs those system wide.

However, in order to again benefit from the community and allow it to contribute to your fork, please fix this issue. There will be no way around.
[image]

<ins>__Some directly related issues - not mentioning issues resulting from the consequence of not supporting Linux build:__</ins>

- https://github.com/CrealityOfficial/CrealityPrint/issues/311
- https://github.com/CrealityOfficial/CrealityPrint/issues/216
- https://github.com/CrealityOfficial/CrealityPrint/issues/183
- https://github.com/CrealityOfficial/CrealityPrint/issues/116
- https://github.com/CrealityOfficial/CrealityPrint/issues/102
- https://github.com/CrealityOfficial/CrealityPrint/issues/3
- https://github.com/CrealityOfficial/CrealityPrint/issues/210
- https://github.com/CrealityOfficial/CrealityPrint/issues/203
- https://github.com/CrealityOfficial/CrealityPrint/issues/84

## Suspicious dependencies: 
- `dep_OpenSSL`
- `dep_GMP`
- `dep_Boost`
- `dep_OpenCV`
- `dep_Qhull`
- `dep_JPEG`
- `dep_FREETYPE`
- `dep_OpenEXR`
- `dep_NLopt`
- `dep_GLFW`
- `dep_Blosc`
- `dep_TBB`
- `dep_GLEW`
- `dep_PNG`
- `dep_Cereal`

## Detailed Error (example):

```bash
$ ./build_linux.sh -d
...

[15/103] Performing install step for 'dep_OpenSSL'

FAILED: dep_OpenSSL-prefix/src/dep_OpenSSL-stamp/dep_OpenSSL-install /home/ihateposerwithborrowedplumes/creality-print/deps/build/dep_OpenSSL-prefix/src/dep_OpenSSL-stamp/dep_OpenSSL-install 
cd /home/ihateposerwithborrowedplumes/creality-print/deps/build/dep_OpenSSL-prefix/src/dep_OpenSSL && make -j24 install_sw && /usr/bin/cmake -E touch /home/ihateposerwithborrowedplumes/creality-print/deps/build/dep_OpenSSL-prefix/src/dep_OpenSSL-stamp/dep_OpenSSL-install
...

Cannot create directory /usr/local//lib/engines-1.1: Permission denied
make: *** [Makefile:378: install_engines] Error 2
make: *** Waiting for unfinished jobs....
Cannot create directory /usr/local//include/openssl: Permission denied
make: *** [Makefile:322: install_dev] Error 2
...

ninja: build stopped: subcommand failed.
$
```

Btw, [OpenSSL 1.1.1w](https://github.com/openssl/openssl/releases/tag/OpenSSL_1_1_1w) is from __Sep 11, 2023__ and __1.x.x LTS ended__ with same date: 16 months ago. Its probably time to jump on the 3.x.x train (see https://endoflife.date/openssl). 

Kind regards!
rr

Liionboy, 2025-02-01:
I am trying to compile with a Intel 12th Gen Intel© Core™ i7-12700H × 14 and 16 GB Ram and always my laptop hang anc can't fininsh, i don't have that error on SSL but eat all resources from my laptop and i don't know what PC need to compile this buil.

Thanks

rubienr, 2025-02-01:
Hi @Liionboy,

I tried to build the deps, and then the slicer from the OrcaSlicher repository (i still can't build Creality's)...

Building the dependencies `BuildLinux.sh -d` made no considerable impact on RAM but when i built the slicer `BuildLinux.sh -s`, having already 6GB RAM used, it soon hit the 16GB limit, then went up to about 31GB and sometimes it hit the 45GB limit. The build utilized all my 12 cores  (x2 threads), hence that amount of memory consumption ¯\_(ツ)_/¯ ? I don't know how mad Ninja is if the OOM killer decapitates `cc1plus` tasks.
However if the `-i` flag helps here or just skips checks i don't know, but you could limit the cores:

```bash
export CMAKE_BUILD_PARALLEL_LEVEL=7
BuildLinux.sh -s
```

[...]
Nice! I accidentally started `BuildLinux.sh -s` again and must realize that it builds all again 🤦‍. But this is another story...

Kind regards

[1] https://cmake.org/cmake/help/latest/envvar/CMAKE_BUILD_PARALLEL_LEVEL.html

Liionboy, 2025-02-01:
So, after run succesfully i have this error with AppImage:

```
adrian@adrian-huawei-d16:~/Downloads/CrealityPrint/build$ ls -ltrh
total 822M
-rwxrwxr-x  1 adrian adrian 2,1M Dec  8  2021 appimagetool.AppImage
-rw-rw-r--  1 adrian adrian  256 Feb  1 18:44 buildinfo.h
-rwxrwxr-x  1 adrian adrian 1,3K Feb  1 18:44 build_appimage.sh
-rw-rw-r--  1 adrian adrian 136K Feb  1 18:44 CMakeCache.txt
-rw-rw-r--  1 adrian adrian  306 Feb  1 18:44 CTestTestfile.cmake
-rw-rw-r--  1 adrian adrian 3,3K Feb  1 18:44 cmake_install.cmake
lrwxrwxrwx  1 adrian adrian   46 Feb  1 20:04 resources -> /home/adrian/Downloads/CrealityPrint/resources
-rwxrwxr-x  1 adrian adrian  298 Feb  1 20:04 AppRun
-rw-rw-r--  1 adrian adrian  23K Feb  1 23:19 CrealityPrint.png
drwxrwxr-x  3 adrian adrian 4,0K Feb  1 23:19 usr
-rw-rw-r--  1 adrian adrian  205 Feb  1 23:19 CrealityPrint.desktop
drwxrwxr-x 25 adrian adrian 4,0K Feb  1 23:26 src
-rw-rw-r--  1 adrian adrian 8,1K Feb  1 23:26 compile_commands.json
drwxrwxr-x  4 adrian adrian 4,0K Feb  1 23:26 CMakeFiles
-rw-rw-r--  1 adrian adrian 2,1M Feb  1 23:26 build.ninja
-rwxrwxr--  1 adrian adrian  298 Feb  1 23:26 CrealityPrint
-rw-rw-r--  1 adrian adrian 554M Feb  1 23:26 Creality.tar
-rwxr-xr-x  1 adrian adrian 264M Feb  1 23:26 CrealityPrint_V6.0.0_Alpha.AppImage
drwxrwxr-x  5 adrian adrian 4,0K Feb  1 23:26 package
adrian@adrian-huawei-d16:~/Downloads/CrealityPrint/build$ ./CrealityPrint_V6.0.0_Alpha.AppImage 
[2025-02-01 23:30:02.511076] [0x00007d883758e480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(process:131060): GLib-GObject-CRITICAL **: 23:30:02.785: invalid (NULL) pointer instance

(process:131060): GLib-GObject-CRITICAL **: 23:30:02.785: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```

Here i don't know what to do but i will try search

Liionboy, 2025-02-01:
Another problem, if i download Source code and try to build is a error from what i see and can't resolv:

`[171/193] Performing download step (download, verify and extract) for 'dep_OCCT'
-- Downloading...
   dst='/home/adrian/Downloads/CrealityPrint-6.0.2/deps/build/../DL_CACHE/OCCT/V7_6_0.zip'
   timeout='none'
   inactivity timeout='none'
-- Using src='https://github.com/Open-Cascade-SAS/OCCT/archive/refs/tags/V7_6_0.zip'
-- verifying file...
       file='/home/adrian/Downloads/CrealityPrint-6.0.2/deps/build/../DL_CACHE/OCCT/V7_6_0.zip'
-- Downloading... done
-- extracting...
     src='/home/adrian/Downloads/CrealityPrint-6.0.2/deps/DL_CACHE/OCCT/V7_6_0.zip'
     dst='/home/adrian/Downloads/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT'
-- extracting... [tar xfz]
-- extracting... [analysis]
-- extracting... [rename]
-- extracting... [clean up]
-- extracting... done
[173/193] Performing patch step for 'dep_OCCT'
FAILED: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch /home/adrian/Downloads/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch 
cd /home/adrian/Downloads/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT && git apply --directory deps/build/dep_OCCT-prefix/src/dep_OCCT --verbose --ignore-space-change --whitespace=fix /home/adrian/Downloads/CrealityPrint-6.0.2/deps/OCCT/0001-OCCT-fix.patch && /usr/bin/cmake -E touch /home/adrian/Downloads/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory
ninja: build stopped: subcommand failed.`

rubienr, 2025-02-03:
Hi @Liionboy,

w.r.t. `BuildLinux.sh -s` that is rebuilding all any time i call it - well that was my bad. The documentation is sparse. After running the script once the build directory is in place and you could:
```bash
cd build
ninja all -d explain
# or alternatively
cmake --build ./ --target all -- -d explain
```
Use `explain` for more details. 

IMHO the help text  of `BuildLinux.sh -h` has plenty room for improvement.

---

- `invalid (NULL) pointer instance`, `g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)` sounds related to GTK. 
- `Performing patch step for 'dep_OCCT' FAILED: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch ...` looks like a patch cannot be applied to the occt dependency

Can you `rm -drf deps build` and start a fresh build:
```bash
BuildLinux.sh -d
BuildLinux.sh -s
```

hth

Liionboy, 2025-02-05:
Hi @rubienr ,

I tryed with DockerBuild and is the same error, i will try as you say to see if work now, i will post here results.

Thanks

Liionboy, 2025-02-06:
Nothing , same error with what you say, i think is a problem in code not at use because i saw on Prusa something similar and they fix after a PR on SourceCode

Thanks

Schildkroet, 2025-02-11:
> So, after run succesfully i have this error with AppImage:
> 
> ```
> 
> (process:131060): GLib-GObject-CRITICAL **: 23:30:02.785: invalid (NULL) pointer instance
> 
> (process:131060): GLib-GObject-CRITICAL **: 23:30:02.785: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
> ```
> 
> Here i don't know what to do but i will try search

Try deleting your config in ~/.config/Creality.
This is a error in reading the json.

Liionboy, 2025-02-11:
> Try deleting your config in ~/.config/Creality. This is a error in reading the json.

I did this but after this i have another error with segmentation fault. Problem is at Creality because they didn't do something in code for compile correctly for linux

Thanks

rubienr, 2025-02-14:
Hello @Liionboy,

Sorry for the late reply. The reason is: i gave up with Creality's OrcaSlicer fork and went back to github.com/SoftFever/OrcaSlicer. If you still are trying to compile multiple times, here a few things i can recommend against frustration:

- [1] disable precompiled headers PCH for deps,
- [2] install ccache,
- [3] and probably enable colored output to better spot compiler errors.

## [1,3] PCH (and colors):

Find the line in `if [[ -n "${BUILD_DEPS}" ]] ; then` in BuildLinux.sh  and add a few lines below `-DSLIC3R_PCH=OFF`
to the `cmake` command (do not be confused by  `buld_linux.sh`, in my source that is equivalent to `BuildLinux.sh`):
```patch
diff --git a/build_linux.sh b/build_linux.sh
index 179faf09a..68900fa34 100755
--- a/build_linux.sh
+++ b/build_linux.sh
@@ -130,7 +130,7 @@ if [[ -n "${BUILD_DEPS}" ]] ; then
         if [ ! -d "deps/build/release" ] ; then
             mkdir deps/build/release
         fi
-        cmake -S deps -B deps/build/release -G Ninja -DDESTDIR="${SCRIPT_PATH}/deps/build/destdir" -DDEP_DOWNLOAD_DIR="${SCRIPT_PATH}/deps/DL_CACHE" ${BUILD_ARGS}
+        cmake -S deps -B deps/build/release -DSLIC3R_PCH=OFF -G Ninja -DDESTDIR="${SCRIPT_PATH}/deps/build/destdir" -DDEP_DOWNLOAD_DIR="${SCRIPT_PATH}/deps/DL_CACHE" ${BUILD_ARGS}
         cmake --build deps/build/release
         BUILD_ARGS="${BUILD_ARGS} -DCMAKE_BUILD_TYPE=Debug"
     fi
@@ -154,8 +154,9 @@ if [[ -n "${BUILD_ORCA}" ]] ; then
     else
         BUILD_ARGS="${BUILD_ARGS} -DBBL_RELEASE_TO_PUBLIC=1 -DBBL_INTERNAL_TESTING=0"
     fi
-    echo -e "cmake -S . -B build -G Ninja -DCMAKE_PREFIX_PATH="${SCRIPT_PATH}/deps/build/destdir/usr/local" -DSLIC3R_STATIC=1 ${BUILD_ARGS}"
+    echo -e "cmake -S . -B build -G Ninja -DCMAKE_COLOR_DIAGNOSTICS=ON -DSLIC3R_PCH=OFF -DCMAKE_PREFIX_PATH="${SCRIPT_PATH}/deps/build/destdir/usr/local" -DSLIC3R_STATIC=1 ${BUILD_ARGS}"
     cmake -S . -B build -G Ninja \
+        -DSLIC3R_PCH=OFF \
         -DCMAKE_PREFIX_PATH="${SCRIPT_PATH}/deps/build/destdir/usr/local" \
         -DSLIC3R_STATIC=1 \
         -DORCA_TOOLS=ON \
```

## [2] ccache

- `sudo apt install distcc`
- ccache config: This step is optional, you can use the default config; here is mine as an example:
    ```ini
    cat ~/.ccache/ccache.conf                                               130 ↵
    max_size   = 10G
    ```
- Watch cache hits with `ccache -svv` while compiling, they will increase the 2nd time you compile. On my machine it reduces a "full compilation" of a fresh build folder to about 2 minutes. Wit distcc in doubt you can safely `rm -drf ./build` and rebuild again with `BuildLinux.sh -s`.

## [3] Coloring

This is not vital but helps a lot.

```patch
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 4016c8c4e..97fa50a50 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -1,5 +1,17 @@
 cmake_minimum_required(VERSION 3.13)
 
+option (FORCE_COLORED_OUTPUT "Always produce ANSI-colored output (GNU/Clang only)." FALSE)
+if (${FORCE_COLORED_OUTPUT})
+    if ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "GNU")
+       add_compile_options (-fdiagnostics-color=always)
+    elseif ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "Clang")
+       add_compile_options (-fcolor-diagnostics)
+    endif ()
+endif ()
+
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fdiagnostics-color=always")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -fdiagnostics-color=always")
+
 if (APPLE)
     # if CMAKE_OSX_DEPLOYMENT_TARGET is not set, set it to 11.3
     if (NOT CMAKE_OSX_DEPLOYMENT_TARGET)
```

Liionboy, 2025-02-14:
Still same error:

`[7/26] Performing patch step for 'dep_OCCT'
FAILED: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch /home/creality/Downloads/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch 
cd /home/creality/Downloads/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT && git apply --directory deps/build/dep_OCCT-prefix/src/dep_OCCT --verbose --ignore-space-change --whitespace=fix /home/creality/Downloads/CrealityPrint-6.0.2/deps/OCCT/0001-OCCT-fix.patch && /usr/bin/cmake -E touch /home/creality/Downloads/CrealityPrint-6.0.2/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory
ninja: build stopped: subcommand failed.`

Liionboy, 2025-02-14:
They release a new version and still they don't care about Linux user, nice

wen422, 2025-05-16:
Creality bot: this issue is stale because it has been open for 90 days with no activity.

closed 2025-05-23 by github-actions[bot]