CrealityOfficial/CrealityPrint #325: Linux AppImage

Liionboy, 2025-02-03:
Hello,

Give us linux version o version 6.0.x , we can't use you slicer because this missing!!!!! Respect Linux user!!!

Thanks

mcwalter33, 2025-02-03:
Was wanting to get the k2 plus but without the slicer there isn't much point

jaredcrean, 2025-02-03:
version 5 still works on linux, thats what I have been using for now even though its buggy as hell. Been meaning to try Orca-slicer with it but have not had a chance. Version 6 has a lot more featrues and fixes, hope they release a build soon.

I dont like my operating system spying on me, Linux for the win!!

Liionboy, 2025-02-05:
I tryed to compile in Docker but i have same error , i don't know why they put a release and can't compile if you want

supra89t, 2025-02-25:
You'd think M$ and Apple pay companies NOT to support free operating systems...

BlueCase, 2025-03-17:
Meanwhile you can try the windows version with wine.
Works quite well..
Flatpak Bottles - Runner Caffe - Windows 7 Compatibility for the webview2 binary

BlueCase, 2025-03-17:
@supra89t 
That I couldn't confirm. AMD 4750G (without any tuning)
Slice take a similar time as with orca (2.2 AUR Build) or creality print 5 (appimage)

And sorry but what you expect... They not released a Linux version but you expect the build is working? Also Orca self have sometimes issues when you build it from source.
They released the source code which fulfill the Licences. (Yes I know also here they are not 100% clean)
Nevertheless.. I also disappointed about that the Linux Version seems to be the always last point on the todo list. But also the mac versions (specially the arm release)  was not provided on the first day.
Somewhere in the official channels they talked about to release a Linux Version in March. So hope they will do it so...

davidmcnabnz, 2025-03-31:
> Meanwhile you can try the windows version with wine. Works quite well.. Flatpak Bottles - Runner Caffe - Windows 7 Compatibility for the webview2 binary

@BlueCase I set up a bottle on Debian with a `caffe-9.7` runner and `Windows 7` config, but when the program runs after installing, the main window shows fine but all dialogs render blank. I don't know what you mean by 'webview2 binary'? I couldn't find mention of that in the Bottle configs. Any clues?

BlueCase, 2025-03-31:
@davidmcnabnz 
open legacy tools -> configuration

[image]

[image]

Use webview2 from the bottles dependency install logic (not let creality print download and install a version)

mfraser, 2025-03-31:
Tomorrow is April, still no sign of it.

Liionboy, 2025-03-31:
They release version 6.1 and still no linux version

Goodolandy, 2025-04-02:
I keep checking the software manager, in the hope of seeing CP6 as a flatpak, but sadly, no such luck.

supra89t, 2025-04-03:
> I keep checking the software manager, in the hope of seeing CP6 as a flatpak, but sadly, no such luck.

I stopped using creality print, and I'm using Orca Slicer with RetroD profiles. Perfect prints each time. I have not converted my K1MAX to the CFS system yet as I don't have a need. However what will make me do it is as soon as Orca supports it. I have ZERO faith in Creality software at this point as they do not GAF about their user base.

davidmcnabnz, 2025-04-08:
@supra89t strong but justifiable sentiments. 

If I were to guess, it would be that in CP6, the developers relied much more heavily on Windows specifics that bring a world of pain when trying to compile for Linux. I've followed their compile-from-source instructions to the letter, but the compilation falls over, in a different place every time.

I've had mixed results with Orca, but your mention of "RetroD profiles" sparks my interest. Got a link?

mfraser, 2025-04-09:
> [@supra89t](https://github.com/supra89t) strong but justifiable sentiments.
>
> If I were to guess, it would be that in CP6, the developers relied much more heavily on Windows specifics that bring a [...]

Looks like it is these [OrcaSlicer Profiles](https://www.printables.com/model/600513-orca-slicer-profiles-for-k1-k1c-k1-max-04-and-06-u)

davidmcnabnz, 2025-04-09:
@mfraser wonderful, tried it, got 2 spotless prints from first 2 attempts. Thanks so much!

supra89t, 2025-05-23:
https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.1/CrealityPrint-V6.1.1.2283-x86_64-Beta.AppImage

mfraser, 2025-05-24:
> https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.1/CrealityPrint-V6.1.1.2283-x86_64-Beta.AppImage

Doesn't work in Ubuntu 24.04 as it depends on libwebkit2gtk-4.0-37 which isn't available, only libwebkit2gtk-4.1-0

BloodyIron, 2025-07-06:
Can we please close this redundant thread? Linux versions have been out for a while now.