CrealityOfficial/CrealityPrint #328: Creality Print 6 on Mac does not recognize Spool Holder filament type

jwcowan, 2025-02-06:
If you are tying to use the external filament spool holder instead of the CFS, once you enter the filament information into the K2 and sync with Creality Print, the filament information (PETG vs PLA for example) is not relfected.  When you print you can tell Creality Print to use the external spool holder, but , in my case, the spool holder is set up for PETG while Creality Print thinks it is PLA.  When the print tries to execute, the printer uses the setting for PLA (lower nozzle and bed temps).  If anyone has any ideas, I'm open and fine with user error.

wen422, 2025-05-08:
Creality bot: this issue is stale because it has been open for 90 days with no activity.

ngzhongcai, 2025-06-09:
I am facing the issue also.
I am unable to set a custom filament profile for the EXT spool.
A number of users are asking for this bug to be recognized by Creality and for it to take priority.
[image]