CrealityOfficial/CrealityPrint #329: CrealityPrint_6.0.2.1574 На вкладке "Принтер" неверно отображается заданная температура стола

vd454, 2025-02-10:
На вкладке «Принтер» неверно отображается заданная температура таблицы. Реальная заданная температура стола "текущая" отображается правильно: ~110 гр., "целевая" - 100 гр.
Иногда появляется индикация включения Корпусной вентилятор 100% при температуре в боксе 25+30 гр, но вентилятор не работает. И наоборот при температуре в боксе более 50 гр. Не запуская вручную вентилятор корпуса, камера включается из-за очень высокой температуры. Ручное выключение и включение запускает корпусной вентилятор.

wen422, 2025-05-12:
Creality bot: this issue is stale because it has been open for 90 days with no activity.

wen422, 2025-05-19:
Creality bot: This issue was closed because it has been inactive for 7 days since being marked as stale.

closed 2025-05-19 by wen422