CrealityOfficial/CrealityPrint #335: You still don't give a f... for linux user right?

Liionboy, 2025-02-28:
So this is a reason to buy from another compay because you don't respect linux user , i don't want from you to release a version from linux but put that Source Code for us to compile, you try to compile every version and have a lot of error , why are you so indifferent about linux user? And answer to us because we need a answer!!!!

supra89t, 2025-02-28:
They released the Linux version for V5, then right after killed us with V6. I'm going to use Orcaslicer as this is a fork from it anyways, seeing as a lot of users are heading that way too I'm pretty sure that the CFS will be made to work with Orca in the future.

galosche, 2025-02-28:
Too bad still no Linux. :(

supra89t, 2025-03-08:
You'll see, we will get support 1 month before V7 comes out, then get screwed again for that entire branch because we will only get V7 when V8 comes out.

joshuabridevaux, 2025-03-13:
Wow I'm glad I found this while browsing the issues.  I was about to pull the trigger on a k2 plus but I'm a pure Linux user and don't own a Mac or PC at all.  I think cfs orca support is not quite there yet from what I read so maybe I'll wait until it improves.

hagbardZ, 2025-03-21:
i use Orca instead of this crap. There is always a recent Linux Version of Orca available.

Goodolandy, 2025-04-03:
@hagbardZ are you able to print directly to your printer from Orca?  I have K1 Max and although Orca can see my printer and show me live video, etc, it fails to send a print directly to the printer.

supra89t, 2025-04-03:
@Goodolandy You have to root your printer, then you can print directly to it.

I rooted and installed fluidd. With that I can see the printer, print to it and monitor the live feed from within orca.

kyrubas, 2025-04-06:
Pure Linux user here too. I've tried compiling the repo to an appimage using the `BuildLinux.sh` script and end up with the ninja command missing a needed file. So I guess Orca it is.
@supra89t how does installing fluidd on the printer interact with updates from firmware Creality? Any pointers?

supra89t, 2025-04-06:
@kyrubas Backup your fluidd settings, factory reset, update, install fluidd and restore settings. That's IF you care about the firmware update. 

Lots of info here:
https://guilouz.github.io/Creality-Helper-Script-Wiki/

kyrubas, 2025-04-11:
On the off chance the devs read this and have a spare moment to read this Reddit post, I think you'll find this a compelling argument for supporting Linux more fully/at all.
https://www.reddit.com/r/gamedev/comments/qeqn3b/despite_having_just_58_sales_over_38_of_bug/

witenite, 2025-06-03:
On the Creality website, and in this github, they have 6.1.2 Appimage for Linux. I am running it right now on Debian Trixie, working well.

BloodyIron, 2025-07-06:
Can we please close this redundant thread? Linux versions have been coming out now...

closed 2025-07-10 by Dai-112100