CrealityOfficial/CrealityPrint #337: zhtop does not apply to all routes

oscar6502, 2025-03-03:
Hi.

zhtop does not apply on all routes, this causes the nozzle to drag on the surface, the same model and configuration was tested with OrcaSlicer and it works correctly

[image]
[image]

oscar6502, 2025-03-03:
This is the GCODE of the example

[obj_1_Cat Treat_PLA_1h18m.zip](https://github.com/user-attachments/files/19055518/obj_1_Cat.Treat_PLA_1h18m.zip)

oscar6502, 2025-03-03:
Only occurs when "Apply gap infill Everywhere" is enabled

[image]

Dai-112100, 2025-07-10:
sorry we don't konw what you ask.

Can you give me more clues？