CrealityOfficial/CrealityPrint #341: 6.0.4.1793 can not find Ender 3V

saperlot, 2025-03-08:
Installed the update to 6.0.4.1793 and the software can not find the printer in "Device" menu. 

- App closed and opened again --> still no printer
- Reinstalled 6.0.2.1574 --> Printer is visible
Printer : Ender 3 V3

ikruusa, 2025-03-09:
I have similar issue with K1C (CrealityPrint 6.0.4.1793 on Win10). There is no any sign about whats going on - no error messages, nothing. There is somekind of empty pane labeled as Group1 in Device view.  In version 6.0.2.1574 my printer was visible and everything worked as expected.

saperlot, 2025-03-09:
Hi, thats not just similar, its exactly the same. 
It seems their testing is literally inexistent. They state that they have fixed some LAN printing things, but instead they broke it completely. Very disappointing.

ikruusa, 2025-03-14:
OK, at the end I saw my printer again with the pre-released v6.0.5.1840 . Initially I didn't see it, I re-selected and Confirmed my printer type and then it appeared on the Device tab. Not sure though that it wasn't somekind of delay and re-selecting is unnecessary. Will test it further.

georgebrou, 2025-03-15:
i cant find the printer even i reselect it in v6.0.5.1840. I had to roll back to print something.why they cant just fix it

Dai-112100, 2025-05-09:
Please check again whether version 6.1.1 is correct. We have fixed this issue.

closed 2025-07-10 by Dai-112100