CrealityOfficial/CrealityPrint #342: Slicer doesnt save settings

CalamitouisJ, 2025-03-10:
When closing and reopening the slicer, the selected printer, filament and process settings all go back to the default options. The settings arent deleted, but it is frustrating having to reselect all 3 every single time I open the slicer.

3 versions ago this problem came about and then 2 releases ago it was fixed. Then, in the last release and current release it once again broke

I have no idea how it was managed to be broken, fixed and then broken yet again in the space of a few months, but please lock this down and fix it

hacs-account8934, 2025-03-10:
It's fixed with 6.0.4

[https://github.com/CrealityOfficial/CrealityPrint/releases/tag/v6.0.4](url)

CalamitouisJ, 2025-03-11:
> It's fixed with 6.0.4
> 
> [https://github.com/CrealityOfficial/CrealityPrint/releases/tag/v6.0.4](url)

As I am currently using  6.0.4.1794 I can assure you it is not fixed.

hacs-account8934, 2025-03-11:
That is strange. For me 6.0.4 is working fine. After I once selected my user profile for my printer, filament profile and process profile, I closed CR Print and when I open it again, my profiles are loaded.

Dai-112100, 2025-05-09:
Could you record a screen video to explain your issue? Ideally, it should include your login information for the Chuangxiangyun homepage.

DerekAlfonso, 2026-10-16:
Bump. I recommend this issue be closed since it has long since been addressed.