CrealityOfficial/CrealityPrint #352: Print v6 (both recent releases) give blank screen on Arm Mac

evlfred, 2025-03-28:
[image] this is all I get, can't move past what I assume is a login screen of some sort

GTorreil, 2025-04-13:
having the same issue except the empty window does not show at all. And the main window becomes unresponsive. Can't print anymore...

hemiao2023, 2025-05-06:
Please provide the system version information?

leokabayashi, 2025-05-08:
Same issue here as @GTorreil  for the latest two version of Creality Print.

Happens on Custom Filament profile Preset Screen and for the latest (launched yesterday) happens with "Send Print" screen as well. Unable to print

GTorreil, 2025-05-10:
@hemiao2023 sorry for the delay. The lastest release seems to be working ! It may be resolved, or there may be a special condition triggering the problem. If it occurs again i'll post here. 

I'm leaving this open as other users appear to have the issue on the latest release. Maybe they can help with more debug data. 

System information : 
MacOS 15.4.1 (24E263)

GTorreil, 2025-05-19:
I managed to reproduce the problem again. I think I understand the STR better. 

0. Have an external monitor plugged into the Mac
1. Open Creality Slicer
2. Move the slicer window to the external monitor
3. Slice a model and hit print. 
4. No print window appears and app is frozen. Need to force-quit.

leokabayashi, 2025-05-19:
Great, @GTorreil . 
That's exactly my workflow, I always use it in the external monitor, so I didn't connect the dots here.