CrealityOfficial/CrealityPrint #356: Linux user reporting what I see for version 6.0.5

dbemowsk, 2025-04-11:
When I downloaded the appimage, I was told about the rendering issues which I am dealing with, but one thing I see is that when I have an object on the build plate and I want to delete it, I should just have to click the object and then hit the delete key.  Sometimes it works and sometimes it doesn't. When I can't delete an object with the delete key, if I click and drag the object just a little bit and then hit the delete key, it will delete.  I can also right click on the object an select deletem and that will delete it every time.

As for the rendering issues, when do you anticipate those getting fixed where this will be fully usable for the most part?

kyrubas, 2025-04-11:
Where did you find/get a 6.0.5 app image? I've only ever seen the 5.1.2 and haven't successfully build it owing to  some ninja dependency that's missing.

dbemowsk, 2025-04-12:
> Where did you find/get a 6.0.5 app image? I've only ever seen the 5.1.2 and haven't successfully build it owing to some ninja dependency that's missing.

I was chatting with a guy on discord who pointed me to this.  I don't know how official it is.  It's very buggy.  Things like screen rendering don't work well and I've found some other issues, but I can slice a model and send it to the printer.
https://github.com/jamincollins/CrealityPrint/releases/tag/v6.0.5-beta1

closed 2025-07-25 by github-actions[bot]