CrealityOfficial/CrealityPrint #360: Remove Linux Support.

lvalen91, 2025-04-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

0.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04, 25.04, and Fedora 42

### Printer

K1 Max

### How to reproduce

Attempt building AppImage

### Actual results

Fail

### Expected results

Success?

### Project file & Debug log uploads

1. OpenCV Patch Failure
	•	File: deps/OpenCV/OpenCV.cmake
	•	Problem: PATCH_COMMAND failed due to incorrect working directory.
	•	Fix: Modified PATCH_COMMAND

2. OCCT Patch Failure
	•	File: deps/OCCT/OCCT.cmake
	•	Problem: Patch directory was incorrect.
	•	Fix: Applied the same chdir workaround as OpenCV to ensure correct patch context

3. FFmpeg Inline Assembly Errors
	•	File: deps/FFmpeg/libavcodec/x86/mathops.h
	•	Problem: shr, sar, and other asm operations failed on modern compilers.
	•	Fix: Replaced ic operand constraint with Jc and casted to (int)

4. FFmpeg Build Failure (Doc System)
	•	File: deps/FFmpeg/FFmpeg.cmake
	•	Problem: Texinfo doc build failed due to removed set_from_init_file function.
	•	Fix: Added --disable-doc to the CONFIGURE_COMMAND

5. YangRTC (MetaRTC) Missing Headers
	•	Files Missing:
	•	src/yangaudiodev/linux/, mac/, win/, android/
	•	YangAudioPlayLinux.h, YangAudioPlayMac.h, etc.
	•	Fix: Extracted headers from [MetaRTC v7.072](https://github.com/metartc/metaRTC), placed them into appropriate folders, and resolved include paths:
	•	Changed <...> to "..." or added include guards.
	•	Flattened src/yangaudiodev/yangaudiodev/* to src/yangaudiodev/*.

6. MetaRTC Libraries Not Found
	•	File: src/video/CMakeLists.txt
	•	Problem: find_library(metartc6) failed (wrong name/version).

7. Missing Decoder Symbols (e.g., decode, init_H264DecCont)
	•	Cause: Decoder functions not included in MetaRTC .a builds.
	•	Fix:
	•	Manually extracted decoder source (YangH264VideoDecoderSoft.c)
	•	Created standalone libyangh264decoder.a
	•	Stubbed missing function av_yang_initExtra() with correct signature:

8. Symbol Conflicts (av_malloc, etc.)
	•	File: yangh264decoder/CMakeLists.txt
	•	Problem: Local decoder build duplicated FFmpeg source (e.g., mem.c).

9. AppImage Runtime Segfault
	•	Cause: GTK layout failures and resource path issues.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Too many outdated dependencies referenced or no longer compatible.

Building from source fails without making significant modifications to various parts. Which increases chances of other build processes failing.

If Linux will no longer be supported by this fork of OrcaSlicer. Then mention this in main GitHub page. Too many “bug fixes” since last released AppImage to ignore.

An AppImage can be created from 6.1.0 or 6.1.1 source but requires modifications to openCV, OCCT, ffmpeg, wxWidgets, metartc/Yangdecoder. However, the AppImage fails to run properly due to various issues, possibly attributed to the aforementioned.

BloodyIron, 2025-04-21:
DO NOT REMOVE LINUX SUPPORT many of us still actually want it.

lvalen91, 2025-04-21:
> DO NOT REMOVE LINUX SUPPORT many of us still actually want it.

I’m one of them, but this developer hasn’t published an AppImage since December and the code provided won’t compile an AppImage.

There have been major changes in OrcaSlicer and recent versions of Creality Print for macOS and Windows. Including support for new Creality Printers.

If they mentioned why an AppImage hasn’t been successfully compiled and published. That’s one thing, continuing to release for other operating systems without a word, that’s a bad sign.

rafaelomodei, 2025-04-21:
I recently purchased an Ender 3 V3 printer, and I was disappointed to find that the current Creality Print for Linux does not yet fully support this model.

I use Linux as my primary OS for creating and managing 3D printing projects — just like many others in the maker and developer community. It's really important that you continue supporting and updating the Linux version, as it's a key platform for a large part of your user base.

Please kindly consider keeping Linux support active and up-to-date. It truly empowers your community.

Thank you so much! 🐧🙏

BloodyIron, 2025-04-22:
I E-Mailed Creality Support and 1 hour ago (before the posting of this comment) their response is:

"We apologize for any inconvenience caused.Creality Print Linux version 6.0 is expected to be updated at the end of this week, so please keep an eye on the website and download the update, thank you!"

(and then they linked to the download URL)

So... maybe they've been quietly trying to fix it this whole time and haven't said anything until they could. Either way, yay!

BloodyIron, 2025-04-22:
> I recently purchased an Ender 3 V3 printer, and I was disappointed to find that the current Creality Print for Linux does not yet fully support this model.
>
> I use Linux as my primary OS for creating [...]

I have a V3 KE which works with Creality Print 5.x, have you tried the v5.x version?

dbemowsk, 2025-04-23:
@BloodyIron, does this mean that they are going to post an Appimage file for linux?  I run xubuntu and am currently using a beta version 6.0.5 that is very buggy, but works to be able to load and send prints to my K2 Plus.

krom, 2025-04-24:
I was planning to buy K2 plus combo, but I guess I'll look at other printers.
Creality Print v5 is chashed at launch, v6 was never released (I downloaded first version before it wad removed), but at the same time, full support for other slicers was never added to the printer.

The community is trying to configure the orca slicer to work with the CFS, but for now Creality is on the way to a proprietary solution.

Very sad.

kyrubas, 2025-04-30:
> The community is trying to configure the orca slicer to work with the CFS, but for now Creality is on the way to a proprietary solution.

@krom, Linux (POP Os) user here. I've used the CFS with Orca for dozens of prints, both multicolor and single color fed by the CFS. It works perfectly fine. The only issue I have is on Orca's end- connecting to the printer through the `Device` tab crashes Orca, so I have to view/manage machine stats through the machine IP on port 4408.

sixtysecondstosmash, 2025-05-08:
I was about to make a feature request for this to be updated for Linux, and in the background I find this post. The last Linux release was at the end of December, and removal of Linux support should not be an option.

While the latest version of CP5.x.x does work with the K1C and CFS, there is some strange error that pops up on the screen. It's in Chinese so I can't decipher it.

There are also some incompatibilities when opening a 3mf file saved from Linux CP 5.1.7 on Windows CP 6.1.1. No matter what you revert on the right rail, the "tab" icons are still yellow... indicating that something is still different. Lastly, the post saying that CP6.x for Linux would be released in 1 week is now 2 weeks old.

> but for now Creality is on the way to a proprietary solution

😞

>  I've used the CFS with Orca (v2.3.1-dev) for dozens of prints, both multicolor and single color fed by the CFS. It works perfectly fine. The only issue I have is on Orca's end- connecting to the printer through the `Device` tab crashes Orca, so I have to view/manage machine stats through the machine IP on port 4408.

Would love to go back to using Orca, and the only reason I'm on CP is because of the CFS so 🤞 they can get it working stable

supra89t, 2025-05-23:
> Would love to go back to using Orca, and the only reason I'm on CP is because of the CFS so 🤞 they can get it working stable

I'm using OrcaSlicer with a K1MAX and an CFS. Its working just fine. 
https://wiki.creality.com/en/k2-flagship-series/k2-plus/3rd-party-slicing 

I loaded that printer profile, changed the mesh and print size to match my K1MAX (300x300x300) and it just worked out of the box. I then installed the helper script and fluid and its printing flawless (so far)

supra89t, 2025-05-23:
They also released a CP linux beta version https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.1/CrealityPrint-V6.1.1.2283-x86_64-Beta.AppImage

krom, 2025-05-23:
I found 6.1.1 beta  appimage, and it's launching for me. It seems that Linux support is returned, but I can’t view the model inside the window

A similar problem was in Orcaslicer and there it was solved by turning off the GPU aceleration (by adding environment variable). I tried to solve the problem in Crealy Print in this way, however, this did not help

sixtysecondstosmash, 2025-05-24:
With the Ocra Slicer appimage (and by extension the CP 6.x appimage) installing `webkit2gtk4.0` seemed to get things working if anyone had issues launching it.

lugipfupf, 2025-05-28:
Had to install `webkit2gtk` on ArchLinux in order to get the 6.1.1 beta app image to work.

lugipfupf, 2025-05-31:
Btw, on Arch, I got 6.1.2 through AUR yesterday.

+1 for closing

Beagon, 2025-06-01:
Also +1 for closing. Have been using 6.1.1 and 6.1.2 on both Manjaro and KUbuntu (25.05). Although it needs to be noted for 6.1.2 to work I had to install the following packages on Ubuntu

`sudo apt install libwebkit2gtk-4.1-0 libusrsctp-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav`

BloodyIron, 2025-06-18:
> Not working here in Ubuntu 25.04:
> 
> ```
> /tmp/.mount_CrealinlJD2U/bin/CrealityPrint: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
> ```
> 
> But also other libraries, like libunistring.2.
> 
> I extracted the contents and noticed that AppRun redefines `LD_LIBRARY_PATH`, so even though I have these libraries installed as snap in some directories, the AppImage won't work.
> 
> Couldn't you have figured a better way of distributing these ? Or at least included those libraries in the AppImage ?
> 
> Any advise on how to make it work ?
> 
> I just bought the machine, and I wanted to try something ...

sudo apt install libusrsctp2

that should get you going as I believe this is what I had to do to solve this for my own Ubuntu 25.04 setup

janpfeifer, 2025-06-18:
I tried that didn't work. I also tried creating a docker with ubuntu 22.04 and after lots of trying it lacks a libusrsctp1.

I also attempted creating a snap (based on core22), but that failed on snapcraft not liking to coexist with docker ...

It's very frustrating ... I just wanted it to work. At least the cloud version works :smile:

closed 2026-04-26 by lvalen91