CrealityOfficial/CrealityPrint #362: Please allow to export / import individual user presets (filaments and processes)

irudi, 2025-04-24:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I always have to re-create all my settings after upgrading from Creality Print 4 to 5 to 6 ... :(
I would at least expect that the settings of the previous version get imported automatically. But it doesn't. And even the manual import works for me (macOS 15.4, Creality Print 5 → 6), see #274

And it is not possible to share good settings or download and install material profiles from the filament vendors. They provide presets for bambulab, but not for Creality Print (because there is no import available!).

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

A dialog where I can manage all my user presets: rename, delete, export, import. 

### Describe alternatives you've considered

I searched for tipps how to do it manually (by editing config files or put preset files into the right folder) - but I failed.

GoDllkE, 2025-05-08:
A huge missing feature that lowers community QoL in those last 2 major versions. This can't be prioritized as **critical** or **highest** (comparting to other issues opened), but at least **_high_** would be fare.

wen422, 2025-05-23:
Upgrading from CP4.x to CP5.x indeed requires recreating presets. However, during the initial installation and launch of CP6.x, a prompt will appear asking, "Migrate user presets from version 5.x?", Clicking ‌"Confirm‌" will initiate automatic migration. You can also manually import presets via ‌"File → Import → Import 5.x Presets‌", as illustrated below:

[image]

You can also refer to this video:https://www.youtube.com/watch?v=CMqt-pm_9zw

irudi, 2025-05-25:
Unfortunately this does not work for me! 
Maybe it's a bug in the macOS version?

Here my presets in 5.17

[image]

If I try to import from 6.1.1

[image]

I would like to be able to export / import individual filament or process user presets.