CrealityOfficial/CrealityPrint #364: Crelity print 6 not remembering presets next time its opened

Slowyata, 2025-05-02:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

Creality Print 6.1.041

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

K2 plus

### How to reproduce

Open creality print 6, select presents such as user presets. Close the app and reopen

### Actual results

All presets chosen are resetto default profiles

### Expected results

creality print 6 remembers when i last chose and reuses the same presets

### Project file & Debug log uploads

[printers.zip](https://github.com/user-attachments/files/20017703/printers.zip)

[Untitled.zip](https://github.com/user-attachments/files/20017707/Untitled.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

After exiting and reopening creality print 6, it doesnt remember the print preset, filament preset, or processing presents i used last.

hemiao2023, 2025-05-06:
You can try our latest version [6.1.1 ](https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.1/CrealityPrint_6.1.1.2274_Release.exe)

Wonkman1, 2025-05-08:
I did try in 6.1.1 today and this problem issue still happens for me.  I have to re-import the json files I need every time I start Creality Print.

wen422, 2025-05-13:
I tested with the latest version of the software (Creality Print 6.1.1) but was unable to reproduce the issue. 
My steps were as follows:
1、Import the user material preset you provided(Inland PLA.json、Inland Petg.json).
2、Choose a user material preset.
3、Restart the software.
Are there any additional steps I should perform?

altmenorg, 2025-05-23:
Actually I found out the problem on 6.1.1...
I have some presets that are based on syste filaments -> they are remembered when openening the software
I have some presets that are based on filaments I created from scratch (custom filaments) -> those are not remembered

Pretty weird but that's it

nodtem66, 2025-10-31:
The issue still exists in 6.3.0.3420.
I guess it is the problem of JSON parsing.

When I created a new preset "Test" from the old preset "Old". 
This was the content of `Test.json` in `%USERPROFILE%\AppData\Roaming\Creality\Creality Print\6.0\user\default\process`
```json
{
    "base_id": "GP004",
    "from": "User",
    "inherits": "Old",
    "is_custom_defined": "0",
    "name": "Test",
    "print_settings_id": "Test",
    "version": "25.3.4.18"
}
```

After restarted the app, the "Test" preset disappeared. But when I changed the line `inherits": "Old",` to `"inherits": "",`, the "Test" preset loaded successfully!

## A quick fix for users
- Open the broken preset in `%USERPROFILE%\AppData\Roaming\Creality\Creality Print\6.0\user\default\process`
- Change `inherits` to empty string. 
```json
"inherits": ""
```
- Copy the parent preset to this JSON file and merge it manually.

Good luck!