CrealityOfficial/CrealityPrint #365: Sermoon V1 Pro

Atkinsrj, 2025-05-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

Latest Beta 6.1.1.2151 also previous stable release

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Sermoon V1 Pro

### How to reproduce

Device-> Scan Add or Manual add with IP does not find printer. Previous V4 and V5 find printer and allow viewing of device details. 

### Actual results

Cannot find Sermoon V1 Printer on Lan

### Expected results

Will find printer on network like V4 and V5 do

### Project file & Debug log uploads

The issue has to do with functionality and not settings

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Goodolandy, 2025-05-15:
Glad to see I am not the only one.  My K1 Max needs a new extruder so I hooked up my Sermoon but could not get the software to see it on the network.  I thought it may have been because I am using the Linux version.  Since you are on Windows, I now know it is the software.

wen422, 2025-05-22:
The current version does not support connectivity with the "Sermoon". Compatibility is expected to be implemented in version 6.1.2, with updates anticipated for release around the end of this month or early next month

closed 2025-05-22 by wen422