CrealityOfficial/CrealityPrint #367: CrealityPrint_6.1.0.2041_Release пропала поддержка русского языка верните пожалуйста

Nizami748, 2025-05-08:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

CrealityPrint_6.1.0.2041_Release 
пропала поддержка русского языка верните пожалуйста

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Была поддержка Русского языка, пропала, стало неудобно пользоватся

Ursus63, 2025-05-09:
Русский действительно пропал. Только из версии 6.1.1.2274. В 6.1 русский есть. Я пока откатился на эту версию. Надеюсь поправят.

wen422, 2025-05-14:
Я проверил, версия 6.1.0 поддерживает русский язык.
Возможно, на вашей стороне возникли проблемы при установке.
Рекомендую попробовать установить последнюю версию 6.1.1.[6.1.1 Release](https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.1/CrealityPrint_6.1.1.2274_Release.exe)