CrealityOfficial/CrealityPrint #370: Crealitty print does not start on ubuntu linux version

knightkronos, 2025-05-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

5.1.7

### Operating System (OS)

Linux

### OS Version

Zorin OS 17.3 x86_64 - based on ubuntu distro

### Additional system information

check image attached

[image]

### Printer

K1 MAX

### How to reproduce

Downloaded latest version on your webpage
gave permision to execute the .appimage file and the application freezes and never starts.

### Actual results

aplication sometimes closes or freeze.

### Expected results

the application freezes and does not respond, a message saying the app is unresponsive giving the options to force quit or wait, if i press wait it comes back again asking about the same

### Project file & Debug log uploads

there is not logs on the $HOME/.config/Creality Print/log

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Goodolandy, 2025-05-15:
Did you try the other Linux file? CrealityPrint-V6.1.1.2283-x86_64-Beta.AppImage
That one works for me.

knightkronos, 2025-05-20:
how do I download that version?

supra89t, 2025-05-23:
https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.1/CrealityPrint-V6.1.1.2283-x86_64-Beta.AppImage