CrealityOfficial/CrealityPrint #374: Upload failed without any other informations Mac

Counterpart5, 2025-05-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

6.1.1.2274

### Operating System (OS)

macOS

### OS Version

15.1.1 (24B91)

### Printer

K1C

### How to reproduce

1. Slice Plate
2. Send Print 
3. Send print 
4. Upload failed

### Actual results

worked till yesterday, deleting and reinstalling app didnt help, version 5.1.7 works fine

### Expected results

the file should be send to printer and the print should start

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Counterpart5, 2025-05-22:
Will do the screenrecording today, where do I find the 3mf ?

Counterpart5, 2025-05-23:
[treasure.3mf.zip](https://github.com/user-attachments/files/20409009/treasure.3mf.zip)
the .3MF file

Counterpart5, 2025-05-23:
https://github.com/user-attachments/assets/af11da69-3ac8-4fe7-86bf-56bae8747b84
the screenrecording

hemiao2023, 2025-05-26:
It seems that there was an error in uploading the file. I need a log file to help analyze the problem
[image]

Counterpart5, 2025-05-26:
[log Kopie.zip](https://github.com/user-attachments/files/20435839/log.Kopie.zip)
here you are =)

Counterpart5, 2025-05-26:
the curl worked, and then the "send" worked as well. 
could you provide me the needed port that i can test if it could be possible my network ?

Counterpart5, 2025-05-27:
okay, i looked up my network. connection is there port is not blocked. curl works but upload failed again 

[image]

[log2.zip](https://github.com/user-attachments/files/20455261/log2.zip)

hemiao2023, 2025-06-05:
Try ping your device to see what the network latency is. If it doesn't connect within 5 seconds, it will return a connection failure.

Counterpart5, 2025-06-06:
[image] so latency is all good i guess

wallenium, 2025-06-09:
Same problem - MacBook Air M1
No entry in log file, tries to upload, stuck at 1%
Curl positive, pores 80 & 9999 UDP available

```
xxx@MacBook-Air-2 ~ % nc -vnzu 192.168.20.59 80
Connection to 192.168.20.59 port 80 [udp/*] succeeded!
xxx@MacBook-Air-2 ~ % nc -vnzu 192.168.20.59 9999
Connection to 192.168.20.59 port 9999 [udp/*] succeeded!
```

Upload to cloud and using cloud print working, but sucks.

Little snitch on Mac, worked at the beginning but now it is not longer. Any help?

bionemesis, 2025-07-27:
Just to add to this, I have seen significant issues with uploading since the latest firmware update. Usually I have to send the job multiple times and/or restart my Hi in order to upload successfully. Prior to the last firmware, I never experienced issues uploading.