CrealityOfficial/CrealityPrint #386: Add a setting to show dialogs after save/export gcode only once

tewald, 2025-06-12:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

It's really annoying to have a dialog shown asking to open the directory every time after save or export a gcode file.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add an option to the app settings, or at least a "don't ask me again" checkbox on the dialog.

leifclaesson, 2025-06-28:
Seconded! I came here to ask for this. Many other dialogs have a "remember my choice" option but not this one. Please add this option so that I can choose not to be asked to open the folder every time I save a file. I already know what folder it went in, because i just told it to put it there. :)

Dai-112100, 2025-07-09:
OK  thanks for your feedback。  

I have recorded it and will plan and complete it in the next version.