CrealityOfficial/CrealityPrint #387: Most of buttons grayed out Ubuntu 22.04

Openwrtfunboy, 2025-06-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

6.1.2.2458

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04.5 LTS

### Additional system information

AMD R5 3600/VEGA 56. No proprietary drivers. X11 and Wayland same behavior 

### Printer

K2 plus

### How to reproduce

Go to edit filament settings. Most settings will be not edible and grayed out. If you will fast switch between list of parameters then it may become edible but not all.

### Actual results

Grayed out buttons in filament settings.

### Expected results

Changing parameters.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20730194/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

ummduh, 2025-06-15:
I have this issue as well on 2 different computers. Both on Linux Mint.

Sometimes they'll be editable, but usually not. Sometimes you can get one or 2 variables to change, sometimes none no matter what you do.

duese97, 2025-07-01:
Still an issue with the pre-release v6.2.0

Openwrtfunboy, 2025-08-12:
@hemiao2023 @Dai-112100 @wen422 @aduh95 @truncgil @BriellaBugs @Psych0h3ad 
Hello developer team! I appreciate your time, but please fix this annoying problem! Almost all Linux user suffer from it!