CrealityOfficial/CrealityPrint #388: Creality Print 6.1.2 won't launch

Gazule, 2025-06-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

6.1.2

### Operating System (OS)

Linux

### OS Version

Linux Mint 22.1 Mate

### Additional system information

CPU i3 / 16 go ram

### Printer

Créality Hi combo

### How to reproduce

Creality Print 6.1.2 appimage won't launch ... Creality Print 5.1.7 appimage Creality Print 5.1.7 works well, despite some translation errors into French. despite some translation errors into French.

### Actual results

Nothing

### Expected results

works well

### Project file & Debug log uploads

xxx

### Checklist of files to include

- [ ] Log file
- [ ] Project file

pascalsavigny, 2025-06-17:
Hi, 

Same issue here.

command:
./CrealityPrint-V6.1.2.2458-x86_64-Release.AppImage

result:
/tmp/.mount_CrealivpwweK/bin/CrealityPrint: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
Desktop: Cinnamon

Best regards
P.

Gazule, 2025-06-17:
Hi,
I'm on Linux Mint 22.1 Mate. I have the libwebkit2gtk-4.1.0 package installed.  What's the solution?
Best regards

Gazule, 2025-06-18:
I'm in the same situation... no log for version 6... while there is one for version 5.

janpfeifer, 2025-06-18:
Doesn't work on Ubuntu 25.04 either, same missing library.

Isn't it supposed to include these dependencies in the AppImage ?

vlad-saling, 2025-06-20:
Same here on vanilla Ubuntu 24.04 LTS.

azenyr, 2025-06-26:
Fedora 42 same problem, same error. Appimages are supposed to include all dependencies, this is clearly a packaging problem by Creality .

Gazule, 2025-06-30:
I just checked... unable to downgrade the package for my distribution.

Gazule, 2025-06-30:
a symbolic link? How does it work? I don't know.

Kevinsky86, 2025-07-05:
> I was doing testing to get Creality Slicer 6.1.2 and Bambu Studio working in KDE Neon (current stable) with Wayland. I was successful in getting them to work with X11. I have an Nvidia GTX 1080 Ti and I was using the 570 driver.
> 
> Looks like Creality Slicer is still using libwebkit2gtk-4.0 while Bambu Studio is using libwebkit2gtk-4.1. I also found that I can use Wayland with the Nvidia 550 driver, but not the 570 driver.
> 
> If you already have libwebkit2gtk-4.1 installed, you can use the following 2 commands to symlink 4.1 to 4.0 and this worked perfectly for me.
> 
> sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
> 
> sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18

Just a response for people getting here through Google.
This simlink workaround works on Mint 22 but only with CrealityPrint-V6.1.2.2458-x86_64-Release. 
The newer CrealityPrint-V6.2.0.2827-x86_64-Release has a new dependency issue with libsoup2 that I haven´t been able to find a workaround for.

I guess we need to wait for Creality to get these appimages in order.

cswalker, 2025-07-07:
I've confirmed this for all app images going back to 5.1.7. 

The Symlink solution does in fact create an issue between libsoup2 and libsoup3.
`sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37`

wen422, 2025-07-08:
We provide two software versions for Ubuntu:
Users on ‌Ubuntu 22 systems‌ can try:
https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.2/CrealityPrint-V6.1.2.2458-x86_64-Release.AppImage
Users on ‌Ubuntu 24 systems‌ can try:
https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.2/CrealityPrint_Ubuntu2404-V6.1.2.2458-x86_64-Release.AppImage

wen422, 2025-07-09:
> [@wen422](https://github.com/wen422) the 24.04 AppImage is not easily discoverable from [here](https://www.creality.com/pages/download-software?spm=..page_11657537.creality_print_1.1) - the single "Linux" link gives the impression it will run on any distro, when in reality it does not. Would it be feasible to link to all existing AppImages in the main Downloads page and indicate which Linux distro/version they work for? Thanks!

That's a good idea and thank you for your suggestion. We will consider optimizing this issue.