CrealityOfficial/CrealityPrint #391: Calculation of the print speed

Nyemrinju, 2025-06-25:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Hi, if I'm reading this correctly, Creality Print evolved from the Cura engine? But I wonder why they removed the speed settings during development? With Cura I enter the main print speed and the software works out the rest. I somehow miss this, as the print speeds are so important and difficult to judge for a less advanced user. 

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

It would be nice if this function were also available in Creality Print. So a selection point ‘Main print speed’ and the software calculates the dependent speeds such as infill, top layer etc.. Just like in CURA.

Dai-112100, 2025-07-09:
You can open advance ​​

[image]

Nyemrinju, 2025-07-09:
Apparently you didn't understand me! I would like the other speeds to be adjusted automatically by the system (as in CURA!) when I enter the main print speed. Then I only have to make fine adjustments afterwards. I know how to reach the speeds via the Advanced slider!