CrealityOfficial/CrealityPrint #394: Linux AppImage - Cant launch - libsoup-2.4.so.1 missing

fenpaws, 2025-07-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

6.2.0

### Operating System (OS)

Linux

### OS Version

Nobara Linux 42 (Fedora 42)

### Printer

K2

### How to reproduce

Download the latest image, run it

### Actual results

No windows is popping up, running via terminal reveals:

```
/tmp/.mount_CrealiZzd2Pm/bin/CrealityPrint: error while loading shared libraries: libsoup-2.4.so.1: cannot open shared object file: No such file or directory
```

### Expected results

It should open and work

### Project file & Debug log uploads

There is nothing since it can't even start

### Checklist of files to include

- [ ] Log file
- [ ] Project file

azenyr, 2025-07-06:
Same error here, doesn't open at all. Literally can't use my brand new Creality Hi printer because all my devices at home are Linux and the appimage won't open... Will try a VM. But please Creality package all needed libs inside the appimage.

Bazzite KDE (Based on Fedora Kinoite) latest stable.

teacuptechsa, 2025-07-07:
Had similar issue with prev version 6.1.2.2458. Was hoping 6.2 fixes it, but still not. Creality_Print-v5.1.7.10514-x86_64-Release still works.