CrealityOfficial/CrealityPrint #405: Calibration

Nyemrinju, 2025-07-11:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Why is calibration now switched on by default? It wasn't switched on before. Please change it back so that you can decide for yourself whether you want to calibrate!

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Why is calibration now switched on by default? It wasn't switched on before. Please change it back so that you can decide for yourself whether you want to calibrate!

Dai-112100, 2025-07-16:
Sorry sir, we will change this option to remember the user's setting in the next version.

Dai-112100, 2025-07-21:
6.2.1 fixed