CrealityOfficial/CrealityPrint #409: AppImage Support Being Removed By Major Linux Distributions (REQUEST: Flatpak)

Xipher-Zero, 2025-07-15:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Currently CrealityPrint is only delivered for Linux using the universal packaging format AppImage. There are some fundamental issues with this packaging format mostly stemming from its dependence on an outdated and unmaintained version of the system library **fuse2**. The developer and maintainer of the AppImage packaging format has refused to adapt AppImage to use fuse3 (Which is maintained and secure) resulting in different linux distributions having to include an unmaintained and insecure version of the fuse2 system librart to retain AppImage support. Recently several versions of Linux have decided to no longer include fuse2 libraries in an effort to increase system and user security. This choice is being actively reviewed for the upcoming Fedora 43/44 release and has already been implemented in some downstream distributions including some of the Fedora Atomic desktop spins (Which I currently use). 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

At this time a better solution for distributing linux compatible binaries in a universal packaging format is to use Flatpak (Often delivered by the Flathub repositories). This would do a number of things which would help your linux support efforts. 

1. CrealityPrint delivered as a flatpak would bring actual universal compatibility to all versions of linux that support flatpak as all required libraries are provided within the sandboxed application environment. 

2. There would be no need to deliver separate versions of the flatpak application as differences between end point devices are abstracted by the flatpak system making it truly universal. 

3. At this time CrealityPrint is being delivered as TWO separate AppImage packages on your GitHub (A general AppImage and an Ubuntu Specific AppImage) which indicates current compatibility issues your developers are trying to work around. Flatpak would resolve this additional effort.

If CrealityPrint was submitted to flathub it would be available to ALL linux distributions from a centralized repository by default and would not require additional hosting capacity be provided by Creality.
Please consider a plan to migrate your linux version support away from AppImage and to Flatpak to both improve application security for your users as well as be able to provide CrealityPrint to more of your userbase with less overall development resources having to be allocated for the Linux packaging efforts. Please let me know if this is something that is being considered and if so what the timeframe for implementation would be.

### Describe alternatives you've considered

Right now the only alternative users like myself have is to completely change distributions which still support AppImage (Although it is unclear how long support will last there) or try to run CrealityPrint within containers. Ultimately neither solution is ideal as it is likely fuse2 will be removed from most Linux distributions within the next handful of years.

ZachrielForever, 2025-07-15:
Flatpak is king. This should be swapped over to it asap.
This will help it still be usable in the future, by many distros.
Appimages have too many flaws, either with security, or updates.

RustCurve164932, 2025-07-15:
I don't own a 3D printer, but still want this to pass!

hemiao2023, 2025-07-23:
[image]

closed 2025-07-23 by hemiao2023

Xipher-Zero, 2025-07-23:
A different Gnome runtime could be used which would prevent the 'GNOME 46 runtime message from being displayed. The flatpak installed and CrealityPrint is working properly on all of my systems including ones where none of the AppImages would load previously due to dependency issues or because of the removal of fuse2. I really appreciate Crealitys support was on getting this request fulfilled!

JaneSmith, 2025-07-31:
I got a notification email today regarding the new 6.2.1 release, and was so happy to see a Flatpak package listed as one of the downloads. This is fantastic progress! Now we just need to see it on Flathub.