CrealityOfficial/CrealityPrint #410: Can't change infill density

dbemowsk, 2025-07-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

6.2.0.2828

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04 (Jammy)

### Additional system information

CPU:
•  Model: Intel(R) Core(TM) i7-9700 CPU @ 3.00GHz
•  Cores: 8 cores (8 physical cores, 1 thread per core)
•  Base Clock: 3.00 GHz
•  Max Clock: 4.70 GHz
•  Architecture: x86_64 (64-bit)
•  Cache: 12 MB L3 cache, 2 MB L2 cache, 256 KB L1 cache

Memory:
•  Total RAM: 31 GB
•  Currently Used: 24 GB
•  Available: 5.1 GB
•  Swap: 2.0 GB (with 1.7 GB currently used)

### Printer

Creality K2 Plus

### How to reproduce

Load a model, go to infill and try to change sparse infill density.  

### Actual results

When I try to change the infill density, I am not able to click in the box to change it from 100%.  If I try to change the infill pattern from rectilineat to grid it pops up  dialog saying that grid infill doesn't support 100% infill and asks if I want to switch to rectilinear.  Clicking yes will switch it, but clicking no is upposed to reset density to default non 100% value, but it will not.  Clicking no just pops up the dialog again.

### Expected results

Being able to change the density

### Project file & Debug log uploads

[radiation_shield_base_v2.3mf.zip](https://github.com/user-attachments/files/21246645/radiation_shield_base_v2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

dbemowsk, 2025-07-16:
Sorry for the bug post.  Here is a video of what it was actually doing to me.
https://youtu.be/LvUxQHdWTag
Turns out though that I had an issue that required a restart of my computer and I checked it after that and for some reason it is now working.  Not sure what caused the issue.  You can close this out.

Dai-112100, 2025-07-17:
NONONO  There are new clues in this video. If you check AI infill and adjust the fill ratio, an error will occur.