CrealityOfficial/CrealityPrint #414: bad z-offset on Creality Print v6.2.1 Beta

picharras, 2025-07-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

6.2.1.2931

### Operating System (OS)

macOS

### OS Version

Sequoia 15.5

### Printer

Creality Hi

### How to reproduce

Download Creality Print 6.2.1.2931, and slice any model. Send to printer and start printing

### Actual results

default z offset is too close and causes issues whether it's PLA or PETG. first layers get wrong even not filament is extruded because it's too close.

[image]

### Expected results

No issues with default z offset

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/21323710/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

rolling back to 6.2.0.2827 printings are working well(no z offset issues)

Horace-Jay, 2025-07-19:
Could you please share the Gcode file and 3MF file?

Dai-112100, 2025-07-22:
I checked the default Z offset value of the HI machine and it is 0.

 This should be a problem with the printer leveling. You can report this problem to the customer service.

closed 2025-07-22 by Dai-112100