CrealityOfficial/CrealityPrint #423: hyper ABS and hyper PETG system profiles for 0.8 and 0.6 nozzles for K1 series

3dpartdesign, 2025-07-22:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

If Creality sells hyper PETG and ABS filaments and if it sells 0.6mm and 0.8mm nozzles for K1 series printers then there should be profiles included for that. I mean system profiles.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to see system profiles for 0.8 and 0.6mm nozzles for K1 series printers and hyper ABS / hyper PETG filaments.

Dai-112100, 2025-07-24:
sorry sir ，We haven't received any news about the new consumables from the project team yet. Maybe you can go to the K1 machine after-sales service to promote this matter.

closed 2025-07-24 by Dai-112100

3dpartdesign, 2025-07-24:
https://store.creality.com/products/unicorn-quick-swap-nozzle-kit-for-k1c-ender-3-v3

https://store.creality.com/products/genuine-e3d-obxidian%E2%84%A2-high-flow-nozzle-assembly

Here you have your hardware which is sold in your firm more than half a year I think. Could you talk with your neighboured departament inside your firm about that or should I talk with them about that?

If I should talk with them then please point me to them.

I think your own hardware should be supported inside Creality Print (for K1 series printers).