CrealityOfficial/CrealityPrint #424: Printer Profile: CR-200B profile

gucio321, 2025-07-22:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I can't find Creality CR-200B in CR Series section of the slicer

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Preset should be there

### Additional context

I'm new to this slicer. I want to set up my printer and it isn't in its official slicer for some reason (?) (correct me if I'm wrong)

Dai-112100, 2025-07-24:
you should to download Creality print V4.3.8

closed 2025-07-24 by Dai-112100

gucio321, 2025-07-24:
@Dai-112100 but can't I use the latest version with all the new features?

Dai-112100, 2025-07-24:
Currently, CR-200B only supports version 4.3.8

gucio321, 2025-07-24:
So maybe I can do this?

gucio321, 2025-08-12:
For anyone who ever needs to use this printer with a modern slicer: https://github.com/gucio321/PrusaSlicer-CR-200B

hpflatorre, 2026-08-27:
For anyone still printing on a **CR-200B**: since Creality has said there is no plan to port it to the 5.x+ line, I re-authored the 4.3.8 definitions in the 7.x (Orca-format) profile schema — machine model, 0.4 nozzle machine, 0.08/0.16/0.20 mm processes and Generic PLA/ABS/PETG filaments, with the retraction, acceleration (Z = 100) and temperature values taken verbatim from the 4.3.8 Cura profiles.

They ship in a maintained fork with a ready-to-run Linux AppImage (repacked from the official 7.2.1 release, same binary):

- Repo / profiles / generator: https://github.com/hpflatorre/CrealityPrint (`resources/profiles/Creality/`, `scripts/cr200b/`)
- Download: https://github.com/hpflatorre/CrealityPrint/releases/latest
- One-line install (Ubuntu/Kubuntu/Mint, needs `fuse3`): `curl -fsSL https://raw.githubusercontent.com/hpflatorre/CrealityPrint/cr200b-kubuntu/scripts/repack/install.sh | bash`

Not opening a PR here given the "no plan" above; the generator (`scripts/cr200b/make_cr200b.py`) can also inject the profiles into the stock AppImage on Windows/macOS-free setups if anyone wants just the JSON files. Caveat: the CR-200B **Pro** (direct drive, 220 mm Z) is a different machine and is not covered.

LuanFaa, 2026-11-10:
Opa, tenho uma CR200B e resolvi essa questao da seguinte forma: Instalei os Presets da Ender 3 V2 e alterei a velocidade e comprimento de retração para 6mm, 50mm/s com desretração de 40mm/s, altura de mesa 200m e dimenções de 200m x 200m, tive problemas com entupimento no começo fazendo troca entre PLA e PETG, entao recomendo ter 2 Hotands para cada material, como ela é fechada e guarda muito calor dentro as peças podem deformar, entao certifiquese que a troca de ar esta ativada sempre, tampas abertas e local refrigerando, caso contrario tem que por um cooler mais potente para sair o ar quente de dentro, o resto é só calibração de acordo com o filamento, mantem a mesma largura de camada como Ex 0,4mm em todos ( mesmo sendo bico 0,2mm), velocidades no maximo 80mms em todos com excessão em Overhangs, que tem que deixar entre 10 a 25 mm/s porque depende do filamento, esse comentario pode ajudar muito usuarios dessa impressora