CrealityOfficial/CrealityPrint #425: Integrate Interface Shells Feature from Orca Slicer

lglasgow, 2025-07-23:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

One of the major items that keeps me from using Creality Print with my K2 Plus over Orca Slicer is the fact that Creality Print has not integrated interface shells into the app.

Here is the wiki article for interface shells:
https://github.com/SoftFever/OrcaSlicer/wiki/multimaterial_settings_advanced#interface-shells

I print a lot if intricate items that have lots of different odd edges and differing heights. The items I print can't be printed down on the build plate due to the differing height issues. Interface shells makes it so that there are less lines etc in the top layers of the print and it gives my prints a better overall look than when not using them. This feature was created for a different purpose but it makes prints look way better on their top layers.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would love to see Creality Print integrate the Interface Layers feature from Orca Slicer.

### Describe alternatives you've considered

I use Orca Slicer for most of my prints due to the fact that it has the Interface Shells feature but I don't get the same access to the printer like I do with Creality Print. Access items like CFS access. Choosing colors from the CFS from the Send Print dialog etc.

Dai-112100, 2025-08-12:
sorry sir We will consider this feature when we make multiple extruders

closed 2025-08-12 by Dai-112100

AshenRabbitStudios, 2025-12-30:
I cannot use Creality slicer for my prints until this is implemented. The feature exists in Orca slicer and works for the K2. Surface quality without this feature for my specific models is horrible.