CrealityOfficial/CrealityPrint #431: Filament Start Gcode not added

Beagon, 2025-07-29:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

v6.2.1.2937

### Operating System (OS)

Linux

### OS Version

Manjaro 6.12.39-1-MANJARO

### Printer

K2 Plus

### How to reproduce

1. Open Creality Print
2. Add an object like a cube
3. ONLY USE 1 COLOR
4. Slice plate
5. Export plate sliced file
6. Open the generated gcode in a text editor
7. Look for ;filament start gcode

### Actual results

`;filament start gcode` is not found.

### Expected results

`;filament start gcode` is found.

### Project file & Debug log uploads

[FilamentStartGcodeTest.zip](https://github.com/user-attachments/files/21495793/FilamentStartGcodeTest.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

The filament start gcode does get added when you use multiple filaments. So when using the CFS it does work.

Dai-112100, 2025-08-12:
As far as we know, filament start gcode only needs to be used when changing colors. Please tell me what is the purpose of using filament start gcode when printing in monochrome.

Beagon, 2025-08-15:
> As far as we know, filament start gcode only needs to be used when changing colors. Please tell me what is the purpose of using filament start gcode when printing in monochrome.

I am using the start Gcode to set which filament I am using for my spoolman filament management system so my filament gets tracked correctly using the integration of moonraker with spoolman.

There are other uses like setting certain variables when using a type of filament that might require different/dynamic pressure advance or something else. 

Orca and Bambu work out of the box, why should creality be any different?

donduce, 2025-09-21:
> As far as we know, filament start gcode only needs to be used when changing colors. Please tell me what is the purpose of using filament start gcode when printing in monochrome.

You can also give different z-offsets for the filaments there, which often makes sense to us the other slicers support it.

Beagon, 2026-01-31:
Just a reminder: This is still an issue in CrealityPrint 7.0.1