CrealityOfficial/CrealityPrint #441: Linux Flatpak - Many times opening the printer details page crashes the app immediately

azenyr, 2025-08-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

6.2.1

### Operating System (OS)

Linux

### OS Version

Bazzite 42 (Fedora 42)

### Additional system information

Desktop PC
Ryzen 9 5900X
RX 6900 XT
32GB RAM
2x 4K monitors at 150% scaling (might be related!!!)

Bazzite 42 (Fedora 42)
KDE Desktop 6.4.3 - on Wayland

### Printer

Creality Hi

### How to reproduce

1. Using flatpak on linux open creality software
2. Try to access printer "Details" page where you see its camera etc.
2.1. This also includes the popup right before sending print, every time it tries to load printer camera view it crashes.
3. Crashes immediately, sometimes completely freezes for a few seconds before crashing anyway.

### Actual results

App crashes immediately or freezes and crashes after a few seconds. Sometimes we are unable to print because the "Send Print" screen opens these details in a popup, crashing the app.

### Expected results

Details load correcly, and we are able to send prints.

### Project file & Debug log uploads

[debug_Tue_Aug_12_09_50_58_2.log.0.zip](https://github.com/user-attachments/files/21732402/debug_Tue_Aug_12_09_50_58_2.log.0.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

Just to note that the UI scaling bug #440 I reported a few minutes ago might influence this, or not, I don't know.
Also note that some times it works, but then it stops working again (starts crashing). 95% of times it crashes.

azenyr, 2025-08-12:
I just want to add that the project doesn't matter, it crashes even if you try to open it with an empty project
just open the app, go to Devices tab > Details (on the printer) > crashes immediately

dernippel, 2025-08-29:
I have the same issue on Linux Mint 22.1 and the flatpak version. Killing the existing process will temporary solve it. I think this is related to #443.