CrealityOfficial/CrealityPrint #451: CP does not work on wayland natively

vovan2055, 2025-09-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

6.2.2

### Operating System (OS)

Linux

### OS Version

Arch rolling

### Additional system information

Display Card: Nvidia GeForce GTX 1060 6GB

### Printer

K1C

### How to reproduce

1. Install the flatpak version of CP
2. Using flatpak permission manager (i.e. flatseal) disallow the X11 socket for CP
3. Try to launch CP

### Actual results

CP does not launch on wayland. When using xwayland instead, the 3D view is blanked out for me

### Expected results

CP works on wayland natively as well as on xorg and xwayland

### Project file & Debug log uploads

flatpak run io.github.crealityofficial.CrealityPrint
[2025-09-06 16:38:56.446956] [0x00007fa5c38b2480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(CrealityPrint:2): GLib-GObject-CRITICAL **: 16:38:56.516: invalid (NULL) pointer instance

(CrealityPrint:2): GLib-GObject-CRITICAL **: 16:38:56.516: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

### Checklist of files to include

- [x] Log file
- [ ] Project file

djtecha, 2025-09-24:
Yea, broken for me too. Tried flatpak and AppImage. All load but the modeling is empty, very frustrating as a new user.

edson-dias, 2025-09-27:
It looks like this problem is related with wayland. Running a session using xorg solved it to me.

3d-blags, 2025-09-27:
I held my breath as the latest beta rolled out.... BUT still broke. Come on guys, Distros are dropping X11 so time to focus on this issue.

PresiMarsopa, 2025-11-09:
I'm having the same problem with both Creality Print and Orcaslicer in Fedora 43 KDE 6.4 with Wayland
PrusaSlicer, Cura, and BambuStudio work perfectly.

Fr0g-Not-Found, 2025-12-14:
> I'm experiencing the same problem. (Arch Linux + hyprland + RTX3060)
> 
> I also tried using OrcaSlicer, but the behavior is the same.
> 
> Except for minor glitches, it runs 'ok' in a kde-plasma(x11) session. But with no prepare or previews available when running in a hyperland(wayland) session.
> 
> this the log:
> 
> ```
> lucas@archlinux:~ $ SDG_SESSION_TYPE=x11 CrealityPrint
> 
> [2025-09-10 23:03:34.077460] [0x00007f850fc41d80] [trace]   Initializing StaticPrintConfigs
> 
> add font of HarmonyOS_Sans_SC_Bold returns 1
> 
> add font of HarmonyOS_Sans_SC_Regular returns 1
> 
> add font of NanumGothic-Regular returns 1
> 
> add font of NanumGothic-Bold returns 1
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.836: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.836: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.836: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.837: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.837: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.837: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.838: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.838: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.839: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.839: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.839: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:34.840: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> 
[... 689 lines cut ...]
> request for resource: /homepage/images/Community/common/buy.svg
> 
> Request received: GET /homepage/images/Community/common/buy.svgdone
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:41.309: gtk_widget_event: assertion 'WIDGET_REALIZED_FOR_EVENT (widget, event)' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:41.369: gtk_widget_event: assertion 'WIDGET_REALIZED_FOR_EVENT (widget, event)' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:42.279: gtk_widget_event: assertion 'WIDGET_REALIZED_FOR_EVENT (widget, event)' failed
> 
> 
> 
> (CrealityPrint:44082): Gtk-CRITICAL **: 23:03:42.306: gtk_widget_event: assertion 'WIDGET_REALIZED_FOR_EVENT (widget, event)' failed
> 
> Closed sockets
> 
> ...
> ```
same issue. I dont know what else to add but an willing to provide additional logs for anyone that knows what they are looking for.  I have the same issues he is having in hyprland, but in kde wayland. f