CrealityOfficial/CrealityPrint #459: Mouse Ear Brim Not Applying

robomopa, 2025-09-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### CrealityPrint Version

6.2.2.3203

### Operating System (OS)

Windows

### OS Version

Windows 11 25H2

### Additional system information

Intel i7-13700HX
Nvidia RTX 4060 Mobile

### Printer

Creality K1 Max + CFS

### How to reproduce

When "Mouse Ear" Brim type selected, it doesnt apply to the slice preview. 

[image]

### Actual results

Part slices and prints without mouse ear brim.

### Expected results

Mouse ear brim should be seen in slice preview and print.

[debug_Wed_Sep_24_11_37_28_9220.log - Kopya.zip](https://github.com/user-attachments/files/22514128/debug_Wed_Sep_24_11_37_28_9220.log.-.Kopya.zip)

### Project file & Debug log uploads

[mouse ear.zip](https://github.com/user-attachments/files/22514118/mouse.ear.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Tragen, 2025-10-07:
Same with 6.3.
With OrcaSlicer and the same settings, it's working.

jeffstark, 2025-11-25:
Came here to report this as well.